calyxir/calyx #1: Syntax class for specifying port on submodule

sgpthomas, 2019-07-18:
Right now I have this nonsense:
```
    (pattern (u:id @ uport:id -> v:id @ vport:id)
             #:with fun #'(connect 'u 'uport 'v 'vport))
    (pattern (u:id -> v:id @ vport:id)
             #:with fun #'(connect 'u 'inf# 'v 'vport))
    (pattern (u:id @ uport:id -> v:id)
             #:with fun #'(connect 'u 'uport 'v 'inf#))
    (pattern (u:id -> v:id)
             #:with fun #'(connect 'u 'inf# 'v 'inf#))
```
It really seems like I should be able to factor out a syntax class that has two forms:
(x:id @ port:id) -> 'x , 'port    and     (x:id) -> 'x, 'inf#
and then rewrite the above in a single line. However, I wasn't able to do this without
needing parenthesis around the syntax class. In other words, I could factor it out, but then I would need to write (define/module  ..... [(sub @ left) -> (out)]   instead of [sub @ left -> out].
I wanted to avoid more parenthesis. I know, blasphemy.

rachitnigam, 2019-07-18:
I've been previously able to do this using [~seq](https://docs.racket-lang.org/syntax/stxparse-patterns.html?q=seq#%28form._%28%28lib._syntax%2Fparse..rkt%29._~7eseq%29%29) and `~optional`.

See [this macro](https://github.com/cucapra/seashell/blob/225fa6915030fdf86a678e3752f964e455656103/tools/array-access-visualizer/let-matrix.rkt#L40) in the array visualizer.

Also, this blog post series is very useful: https://www.greghendershott.com/fear-of-macros/

sgpthomas, 2019-07-19:
Got it! I needed a splicing syntax class and some ~seq's and ~optionals

commit 6bba201b (2019-07-19): Fixed #1 !!
closed 2019-07-19 by sgpthomas