calyxir/calyx #3: Use contracts everywhere!

sgpthomas, 2019-07-18:
I don't use contracts in any of my structures. I probably should? I just haven't bothered to look into how they work yet. I'll do that soon.

rachitnigam, 2019-07-18:
See [contracts on structs](https://docs.racket-lang.org/guide/contracts-struct.html) and `define/contract`. Start with `struct`s and move on to commonly used functions.

sgpthomas, 2019-07-24:
@rachitnigam 
I want to do something like this:
```racket
(define ast-node?
  (or/c
   par-comp?
   seq-comp?
   deact-stmt?
   if-stmt?
   ifen-stmt?
   while-stmt?))

;; type of statements
(define-struct/contract par-comp
  ([stmts (list/c ast-node?)])
  #:transparent)
(struct seq-comp (stmts) #:transparent)
(struct deact-stmt (mods) #:transparent)
(struct if-stmt (condition tbranch fbranch) #:transparent)
(struct ifen-stmt (condition tbranch fbranch) #:transparent)
(struct while-stmt (condition body) #:transparent)
```
Is this possible? (I have only used `define-struct/contract` on the first one, pretend they are all there)

rachitnigam, 2019-07-24:
Yeah, I think that should work fine. Careful when exporting the structs. Only files that import structs have their contracts checked, not the ones where you define them.

Read the contracts page to see how they work.

One additional thing to consider is putting minimal efforts in these. The racket version is probably only for prototyping the ideas. We might eventually move to a new language or back to Scala when we start building the RTL backend next semester.

Contracts are generally useful for quick debugging but something like switching to typed racket won’t be.

sgpthomas, 2019-07-24:
Are you sure about the first point?
```racket
(define-struct/contract test
  ([a string?]
   [b number?]))
(test "hi" 4)
(test 4 "hi")
```
this errors immediately. My understanding is that structs defined with `define-struct/contract` get their contracts checked on creation, no matter where they are created from.

sgpthomas, 2019-07-24:
also, the above doesn't work because the par-comp?, seq-comp? etc. functions aren't defined

rachitnigam, 2019-07-24:
Ah, I was talking about the `struct-out` function: https://docs.racket-lang.org/guide/contracts-struct.html?q=struct%20contract

rachitnigam, 2019-07-24:
Consider using `contract-out` or `struct-guard/c`. Not sure why define-contract/struct is not allowing for recursive definitions.

closed 2019-07-26 by sgpthomas