calyxir/calyx #6: Consider using animations instead of GUI

rachitnigam, 2019-07-19:
It might be easier to have a simple looping animation to show all of the computations instead of a GUI (or maybe have both for debugging!)

Here is a simple example with all the code to make a GIF from an input set of images. It uses the [animation functions](https://docs.racket-lang.org/pict/Animation_Helpers.html?q=animation#%28def._%28%28lib._pict%2Fmain..rkt%29._fade-pict%29%29) from the pict library.

```racket
#lang racket

(require pict
         mrlib/gif
         slideshow/play)

(define (do-fade n)
  (fade-pict n (rectangle 30 30) (disk 30)))

(write-animated-gif
  (for/list ([n (in-range 0 1.2 0.2)])
    (pict->bitmap (scale (do-fade n) 3)))
  30
  "animated.gif"
  #:loop? true)
```

sgpthomas, 2019-07-24:
Eventually I might do this, but at the moment a lot of the work is handled by the `mrlib/graph` library. They handle the arrows and the layout.

sgpthomas, 2019-08-08:
I ended up doing both and doing the animation in the gui. It's nice to have stepping for debugging purposes.

closed 2019-08-08 by sgpthomas