calyxir/calyx #8: Better logging system

sgpthomas, 2019-07-24:
@rachitnigam Do you have experience with the racket logging system? I have a janky implementation in `util.rkt` atm, but I'm sure there is a better way.

rachitnigam, 2019-07-24:
Unfortunately not al lot. I usually just use the pln style function for debugging. 

A truck worth looking into is parameters. Parameters allow you to control the dynamic binding of some variable in a thread using parameterize. Define a parameter debug? and set it to false. Next define a macro that only logs when the parameter is true. When you want to debug a piece of code, run it by wrapping it with parameterize debug? true.

Parameters are another one of those crazy powerful racket composition mechanisms. Turns out you can have good abstractions without types.

rachitnigam, 2019-07-24:
Here's a minimal example of this in action:

```racket
#lang racket/base

(require racket/pretty)

(define should-print? (make-parameter #f))

(define-syntax-rule (debug v)
  (begin
    (when (should-print?) (pretty-print v))
    v))

(define (compute x)
  (+ (debug x) 1))

(compute 10) ; Doesn't print anything

(parameterize ([should-print? #t]) ; print the input to compute using debug macro.
  (println (compute 20))) ; println added to print final result
```

sgpthomas, 2019-08-08:
@rachitnigam I figured out a slightly more built-in solution to this. check out [with-logging-to-port](https://docs.racket-lang.org/reference/logging.html#%28def._%28%28lib._racket%2Flogging..rkt%29._with-logging-to-port%29%29). It's more or less what you did but using loggers and the with-logging-to-port form instead of parameters.

closed 2019-08-08 by sgpthomas

sgpthomas, 2019-08-08:
the `with-logging-to-port` only works when you don't have threads :(

rachitnigam, 2019-08-09:
Oh that’s weird. Example?

Also, nice! I was thinking what it would take to implement your system with threads.

rachitnigam, 2019-08-09:
Parameters are a thread concept so it should work with threads fine.

sgpthomas, 2019-08-09:
I was rushing and was mistaken about the threads. I was just using the `with-logging-to-port` thing wrong. Here's a working example. It definitely is more complicated, but it gives you a little more control because you can create hierarchies of loggers and can filter on topics.
```racket
(require racket/logging)

; define a logger called 'test-logger
(define test-logger (make-logger 'test-logger))

; log a message to it on topic1, nothing prints
(log-message test-logger 'debug 'topic1 "some message!!")
; log a message to it on topic2, nothing prints
(log-message test-logger 'debug 'topic2 "some message!!")

; capture logging to test-logger at 'debug level on topic 'topic1
; and send to stdout
(with-logging-to-port (current-output-port)
  (lambda ()
    (log-message test-logger 'debug 'topic1 "some message!!")  ; prints
    (log-message test-logger 'debug 'topic2 "some message!!")) ; doesn't print
  #:logger test-logger
  'debug
  'topic1)

; thread that waits to receive a message and then logs it
(define thrd
  (thread (lambda ()
            (let loop ()
              (define msg (thread-receive))
              (log-message test-logger 'debug 'topic1 msg)
              (loop)))))

(thread-send thrd "un msg")    ; nothing prints
(thread-send thrd "otra msg")  ; nothing prints

; wrap the thread-sends in with-logging-to-port
; this is what I thought originally didn't work
(with-logging-to-port (current-output-port)
  #:logger test-logger
  (lambda ()
    (thread-send thrd "un msg")     ; prints
    (thread-send thrd "otra msg"))  ; prints
  'debug)
```

sgpthomas, 2019-08-09:
Here's an example of the hierarchy: (note you need >=racket-7.1 for the define-logger form to work with parents)
```racket
; define a chain of 3 loggers
(define-logger base)
(define-logger ast #:parent base-logger)
(define-logger compute #:parent ast-logger)

(with-logging-to-port (current-output-port)
  (lambda ()
    (log-ast-debug "some fmt ~v" '(hi bye))) ; outputs because base is higher than ast
  #:logger base-logger
  'debug)

(with-logging-to-port (current-output-port)
  (lambda ()
    (log-ast-debug "some fmt ~v" '(hi bye))) ; doesn't output because compute is lower than ast
  #:logger compute-logger
  'debug)
```

commit 3a10718c (2019-08-08): Improved logging, closes #8