calyxir/calyx #23: Update readme to be accurate

sgpthomas, 2019-10-07:


kwf37, 2019-10-07:
With regard to semantics, I have a few things:

- I would specify that a component has two possible states, "enabled" or "disabled," rather than saying "if it is not active, it is inactive"
- I propose we use the wording that a component can be either "enabled" or "disabled," and that a wire can be either "valid" or "invalid" just to keep them clear. Thus it follows that Enable and Disable statements operate only on components, and wires may be invalidated as a consequence.
- I think the control semantics are otherwise what I expected. Just to rephrase my understanding, our "base cases" are when we actually call "step" and simulate one time step, and otherwise it will make recursive simulation calls. to perform the full simulation.
- Next step would be to define "merge." I think it's the same whether we use it in simulation or step, but may be worth double checking.

sgpthomas, 2019-10-07:
I agree with all these points. About the control, yes your understanding is correct. That section definitely needs to be fleshed out. Also, lets start a branch with the semantics and then we can directly comment on the text that changed. I think it will be easier to keep track of things.

rachitnigam, 2020-03-11:
This issue seems very old. Either remove it, update it, or address it.

closed 2020-03-11 by sgpthomas