calyxir/calyx #34: Library Declaration Proposal

sgpthomas, 2019-10-09:
I don't know if this is the best idea, but...

Maybe we should introduce syntax for library declarations that declare the interface for primitive components:

The reason to do this explicitly so that we have a well defined source where primitives are defined that is not just documentation + implementation, and maybe this would also be useful if different backends supported different sets of primitives.

The syntax would look something like this:
```
(library
  (declare-component (name params ...)
    ((port in-a 32) ...)   ; inputs
    ((port out-a 32) ...) ; output ports
   ; maybe more flags, like whether it increments the time stamp
  )

  ...
)
```
@kwf37 Let me know what you think

kwf37, 2019-10-09:
Yes I like this notion.  One question is what can the params modify? My first thought is that params should only be used to modify the interface (i.e. multiported memories, parameterized bitwidths) and not have anything to do with implementation details, but if we wanted we could have some primitives where parameters adjust loop unrolling factors or something to increase performance.

kwf37, 2019-10-09:
```
(library
  (declare-prim 
    (name (param_1 default_value) (param_2 default_value) ...)
    ((port in-a param_1) ...)   ; inputs (Can be parameterized on params)
    ((port out-a 32) ...) ; output ports
    (arr (port read 32) param_2) ; Perhaps some way of generating multiple input/output ports?
    (sequential true) ; True only for regs for now
  )

  ...
)
```

sgpthomas, 2019-10-09:
Sorry, these thoughts are a bit scattered:

I'm hesitant to have too many restrictions on what parameters can do, mainly because this would be harder and unnecessarily complex to implement. The idea is more that this is a very general interface + some metadata, not a new language for specifying library components.

The places we're using parameters now is for constant modules and for the size and dimension of memories. Bit-widths are the next natural thing to parametrize. Memories are a bit strange because I've been specifying them like this: `(comp/memory 4 8 6)` which would be a 3d array of size `3 x 8 x 6`. But in general this means primitives can take any number of params? Specifying this in syntax seems weird and hard. Maybe we don't allow infinite params? and just have a different primitive for each memory size, and hope people don't make 10 dimensioned arrays. Or maybe we don't specify the params at all in this library thing. This seems like the wrong thing to do, this information is probably helpful for the backend.

Do FPGAs with multiported memories have both multiported memories and non multiported memories? I got a sense from somewhere, but I'm likely wrong, that either all of your memories are multiported or none of them are. In which case it doesn't seem like primitives are the right place to parametrize this. It's more like we want to parameterize the backend on whether there are multiported memories or not.

kwf37, 2019-10-09:
Not sure about the multi ported memory thing. I'll think about this stuff after my prelims and get back to you with some more thoughts.

sampsyo, 2019-10-12:
> I got a sense from somewhere, but I'm likely wrong, that either all of your memories are multiported or none of them are.

On this small matter: in general, both Xilinx and Intel FPGAs seem to have generic memory tiles that can be configured as either single- or dual-port. [This page in the Intel docs](https://www.intel.com/content/www/us/en/programmable/documentation/mtr1422491996806.html#mtr1422491978624) sheds a little bit of light on what these memory modes are, for example. The only thing is that I'm not sure why you would ever use the single-port configuration when the dual-port version is available—can't you just _not_ use the other port? There must be some resource advantage to doing so.

sgpthomas, 2019-10-14:
To pick this up again, I think that the syntax that we should introduce is this (modulo some renaming):
```lisp
(library
 (define/prim (add B)
   ((port left B) (port right B))
   ((port out B) (port carry 1)))

 (define/prim (memory1d word-size addr-width port-mode)
   ((port data-in word-size) (port addr addr-width))
   ((port out word-size #t)))

 (define/prim (register B)
   ((port in B) (port res 1))
   ((port out B #t))))
```
In English, library primitives can declare parameters which can then be used in the port definitions to specify bitwidth. A primitive can also specify parameters that are not used as bitwidths. These can be used by an implementation of the primitive but we will not specify how. 

Calyx will use these these declarations to do a "typechecking" / wellformedness pass to make sure that components are wired up in valid ways. Valid means that port names actually exist and port widths between connections match.

I've also included this `#t` on some of the output ports. This indicates that this output port increments the timestamp on that wire. I think that it makes the most sense to do this on the port level rather than the component level because some components might only want to increment the timestamp on some of their outputs. 

With regards to the dual port, I guess it makes sense to just include this as a parameter for now. However, I think that this may require more thinking and there is no reason to decide this at part of this proposal. 

To layout the things that need to be done to implement this:
* parser for this new language fragment
* make a structureless, controlless component type (maybe just make those fields optional)
* implement a wellformedness pass, maybe just as a part of constructing the structure graph
* update the Dahlia backend to generate bitwidths

kwf37, 2019-10-15:
Sounds good! I'm just a little worried about #t and associating it with ports, but we can see how it goes.

sgpthomas, 2019-10-15:
whats your worry?

kwf37, 2019-10-16:
Will the primitives files still be built-in libraries that they can import to use when compiling? Since each primitive library will have some kind of associated backend, can we setup a file structure where each library file is in the same folder as its associated backend-specific RTL files?

sampsyo, 2019-10-16:
A super cool way this could work would be that something like `define/prim` could be backed _either_ by RTL files or by more Futil code. That is, these declarations would work just like a "header" that describes otherwise-black-box RTL to bring it into the system. Reducing the distinctions between Futil and non-Futil hardware would be aesthetically pleasing.

Separately, you could totally imagine using this infrastructure even when the functionality is not backend-specific. That is, you might want to write some blocks in plain RTL for ultimate low-level control and wrap that up in your Futil design for ease of use.

closed 2020-05-26 by sgpthomas