calyxir/calyx #43: Workshop Conference List

kwf37, 2020-01-14:
Thread for figuring what workshops to submit Futil/Calyx to.

sampsyo, 2020-01-14:
- I keep recommending CAL, which is a journal for early-stage work: https://www.computer.org/csdl/journal/ca
- I didn’t see anything perfect in the ASPLOS workshops: https://asplos-conference.org/co-located-events/
- Nothing really relevant in the PLDI workshops, but that’s not exactly a surprise: https://conf.researchr.org/home/pldi-2020
- I don’t believe the lists of workshops are out for ISCA yet (and they may not be for a little while?), but we can keep an eye out.

sgpthomas, 2020-01-16:
- Probably way to hardware focused but about the right deadline: [https://www.computer.org/digital-library/magazines/mi/call-for-papers-special-issue-on-chip-design-2020](https://www.computer.org/digital-library/magazines/mi/call-for-papers-special-issue-on-chip-design-2020)
- Seems relevant, accepts 4 page WIP papers, but probably too soon of a deadline (Feb 28) [https://conf.researchr.org/home/LCTES-2020#Call-for-Papers](https://conf.researchr.org/home/LCTES-2020#Call-for-Papers)

sampsyo, 2020-01-16:
Huh, yeah, LCTES's topics are a little broader than I was expecting. We don't have anything to say about embedded systems but it seems like there are at least a few relevant people on the PC. We should consider it if we think we can get something good done in a few weeks!

sgpthomas, 2020-01-21:
[https://conf.researchr.org/track/pldi-2020/pldi-2020-Student-Research-Competition](https://conf.researchr.org/track/pldi-2020/pldi-2020-Student-Research-Competition) possible but not really a "standard" paper format and only allows a single author

sgpthomas, 2020-01-21:
w.r.t to Cal, they accept papers on a rolling dealine with [these](https://www.computer.org/publications/author-resources/peer-review/journals) categories. I don't exactly know which topic Futil would fall under, but maybe "Computers" or "Software Engineering". Computers accept short 8 page papers (with references) and Software Engineering accepts papers of any length less than 12.

sampsyo, 2020-01-22:
- Individuals do enter the SRC, but this doesn't preclude any other form of "publication" (in the sense it's not really a publication; more of an in-person competition that you have to get into the first round of by writing something).
- I think what you're seeing there is a list of all the journals that the IEEE Computer Society offers. The second one on the list is CAL. Here is the [author instructions for CAL](https://www.computer.org/csdl/journal/ca/write-for-us/15055?title=Author%20Information&periodical=IEEE%20Computer%20Architecture%20Letters), which says it's four pages.

closed 2020-03-10 by sgpthomas