calyxir/calyx #48: Loan System Proposal

sgpthomas, 2020-01-21:
This proposes a system of statically "loaning" components to other components. This enables component reuse while providing a way of statically checking if the loan is legal. It also acts as a sort of dual to inlining, where a component can swallow another component for some amount of time and control the stepping of that component.

We propose both "caller" and "callee" syntax:

callee:
```lisp
(define/component foo
  ([left 32] [right 32])
  ([out 32])
  ([X ([in 32]) ([out 32]) ()]) ; take some component matching this signature and call it X
  ( ; structure can use X as a component )
; control can enable X
)
```

caller:
```lisp
(define/component bar
  ...
  (; structure
   [new foo1 (foo std_id)] ; pass std_id as the loaned component to foo when instantiating
   ...)
...)
```

A component is only loaned when the component that it is passed to is enabled. A component can not be loaned to multiple components in the same logical time step.

This proposal enables global memory reuse as well as a way to make FSM generation not break the semantics of Futil.

sampsyo, 2020-01-22:
Woohoo! Yes, this is a great next step. As @sgpthomas and I briefly discussed today, I think we should do this *without* static checking at first. We can build a dynamic checker into the interpreter, which shouldn’t be too hard. Then later we can reconsider what a conservative static check might look like—but it could be quite tricky.

rachitnigam, 2020-07-10:
@sgpthomas close this if the groups proposal overrides this. The one unsolved mystery is how components can loan structure to sub-components.

closed 2020-07-15 by rachitnigam