calyxir/calyx #49: Reducing the size of generated FSM

tissue3, 2020-02-06:
The verilog file we generated is very large. It is because of two issues we might want to optimize for:

First is compiler analysis to remove redundant components. For example, 
```
let x: bit<32> = 0;
---
x := 1;
---
x := 2;
---
x := 3;
```
only needs to create structure for the last component.

Another issue is
```
let a = 10;
let b = 1;
```
creates 3 FSMs:
```
     (par
      (enable a0 const0)
      (enable b0 const1))
```
We actually want merge the 2 assignments to 1 component and enable it, which only creates 1 FSM.

sampsyo, 2020-02-07:
Cool! Both of these seem like targets for running "in-Futil" optimizations of the normal, classic compiler optimization variety. That is, the first one is amenable to dead code elimination, and the second one seems quite simple: it could just allow merging parallel enables. (Unless I'm missing something and `(par (enable x) (enable y))` is not guaranteed to always be equivalent to `(enable x y)`?)

rachitnigam, 2020-07-10:
This is quite broad and stale since the recent rewrites using inliner (#139). #140 is a better tracker for the various optimizations we can do to make the generated verilog better.

closed 2020-07-10 by rachitnigam