calyxir/calyx #50: Rough CSP-style Futil Proposal

sgpthomas, 2020-02-11:
As we discussed last time, it seems clear that we need something like this. I'm not entirely happy with the following proposal, but at least it's something concrete to work with.

The idea is to change the semantics so that each component is always 'running' in a loop. Each component would have two special `channels` named something like `start` and `done`. Each component could additionally declare some number of named `channels`. We would introduce two new control primitives for dealing with channels, `(send <channel>)` and `(recv <channel>)`. `send` waits until a component is `recv`ing on the other end of the channel. similarly for `recv`. For now, I think channel connections should be statically declared (presumably in the structure).

By default, a components control will be wrapped in a loop surrounded with `send`, and `recv`. So
```lisp
(<control> ...)
```
becomes
```lisp
(loop
  (seq (recv start)
          (<control> ...)
          (send done)))
```

Components additionally could manage the `start` and `done` signals explicitly to enable more granular control over the behaviour of the component.

sgpthomas, 2020-02-11:
A concern for this is that there doesn't seem to be any difference between `send` and `recv`. So maybe we just have a single control called `sync`. Another thing is that these channels only have control signals. For me, it is still unclear exactly how a component would communicate that it received a particular bit of data. 

It's also unclear to me how this would work with completely synchronous logic. If you knew the latency of every sub-component + CSP-style for sub-component synchronization, would this be enough to generate 'sound' synchronous Verilog?

sampsyo, 2020-02-13:
Seems reasonable to me! But any particular reason why channels _shouldn't_ be able to contain data? That is, you could need to do `(send <chan> <port>)` to send an actual value from a current port on the channel, and similarly `(recv <chan> <port>)` to get a value from a channel and put it onto a wire. It could be convenient (for `start`/`done` or `sync`) to optionally be able to ignore the value part, but in general it seems like it might be useful to communicate values?

kwf37, 2020-02-13:
With this representation, is there a way for a component's logic to check whether another component is ready or not? For example I might imagine something parallel like a multi-core processor may have a subcomponent for each core, and it may want to balance the workload based off which processors are busy.

Also, +1 on being able to communicate values. I think it makes sense to have the option of starting components with values in one expression instead of maybe having to send sync/start messages and value messages separately.

I'm not sure if this is something we want to represent,

closed 2020-09-08 by sgpthomas