calyxir/calyx #52: Implement HLS scheduling

sgpthomas, 2020-03-10:
We need to demonstrate the viability and expressibility of Futil by showing that it can represent traditional HLS scheduling. We should implement this as a series of passes to lower general Futil control into a _global schedule_.

A _global_schedule_ is a subset of Futil control that only has a top level `seq` and `enable`s.

The way to attack this is to implement passes for each Futil control construct that removes it and replaces it with `(seq (enable A ...) ...)`.  Once we have a form of Futil with only `seq`s (arbitrarily nested) and `enable`, we can flatten the schedule.

tissue3, 2020-03-19:
How should `while` be converted to `seq`?
One solution to this is we can replace `while` with `for`

However, I don't understand why the control struct should only have sequence of enable.

sgpthomas, 2020-03-19:
This is a good question that I don't exactly have the answer for. One option, is to move the body of the loop into a component and turn the `while` into a single `enable` of the new component. The component would be "done" when the while loop is finished running.

tissue3, 2020-03-19:
How will this `while` component be represented? It would be hard to represent in structure because `while` do need control. This only option left is to represent `while` with some embedded design like FSM. I am not sure if that's a good idea since we are touching backend at a higher level.

sgpthomas, 2020-03-19:
I think that this is necessary if we want to compile netlists and/or a control free subset of Verilog (which I think we do). For these backends (and any other backend without primtitive loops), we will need some kind of structural representation of while loops within Futil.

rachitnigam, 2020-07-27:
Subsumed by #153.

closed 2020-07-27 by rachitnigam