calyxir/calyx #70: If to Enable pass can result in data harzard

tissue3, 2020-03-19:
Current implementation of RemoveIf pass can result in data hazard. This pass changes 
```
(if (@ gt0 out) (gt0 a0 const2)  [condition line]
         (enable y0 const3)  [true branch]
         (enable z0 const4)))))  [false branch]
```
to
```
(enable gt0 a0 const2 y0 const3 z0 const4)))
```
However, if some register, say `a0`, is enabled at both the condition line and true/false branch, then we cannot what value is updated to a0. 
The solution to this can be 1) change the one step enable to two steps. 
However, the fundamental problem is, 2) registers have both read enable and write enable. We should either assume read is always enabled, or create split the enable semantic. 

To fix 1): We should Without backend, at the high-level futil, we should assume RemoveIf pass always change if to two steps, not one.
For 2): This actually depends on backend. 
For example, MIPS32 has half clock cycle read and half clock cycle updates. Then merging two steps into one is no longer a problem as long as gt0 has a latency smaller than half of the clock cycle.
Or for register read and components that can be also enabled, we can always assume no control signal goes to them.

sampsyo, 2020-03-20:
About 2): that's a good point that it's typically possible to read and update a register in "half cycles." Maybe we should just make this part of the primitive register module's semantics—that a simultaneous read/write works and the read gets the "old" value?

tissue3, 2020-03-20:
> About 2): that's a good point that it's typically possible to read and update a register in "half cycles." Maybe we should just make this part of the primitive register module's semantics—that a simultaneous read/write works and the read gets the "old" value?

I still feel this depends on backend. RISC-V does not have half cycle updates. There will be many interesting specified passes after backend is specified, e.g. MIPS32+FSM implementation. We can even remove `enable `semantics for all components except registers, e.g. adder always perform addition. I feel such kind of passes can be independent of common optimization passes.

rachitnigam, 2020-03-21:
@tissue3 did #78 fix this?

rachitnigam, 2020-07-09:
Futil has been redesigned to use go/done holes and groups in #139.

closed 2020-07-09 by rachitnigam