calyxir/calyx #75: simple automatic parallelization through distinct sub-graph detection

sgpthomas, 2020-03-19:
This pass would detect enable sub-graphs that are specified to run in sequence, but can actually be run in parallel without changing the observable behavior of the program. For example, consider the following program:
```lisp
(seq (enable A B)
     (enable C D)
      ...)
```
Suppose that there are no edges between the sub-graph induced by `(enable A B)` and the sub-graph induced by `(enable C D)`. If this is the case, then there is no way for these subgraphs to depend on each other and then can be run in parallel. We can transform this program into:
```lisp
(seq (enable A B C D)
      ...)
```
There are certainly more complicated cases where you might be able to parallelize the control, but for this pass let's focus on this one simple case.

tissue3, 2020-03-19:
Shall we add some data dependency analysis, or is it enough to check sequence of enable have common components?

sgpthomas, 2020-03-19:
I think some data dependency analysis is needed to make this sound. I don't think it's enough to just check common subcomponents. I think you also have to make sure there are no data edges between the sub-graphs.

tissue3, 2020-03-19:
I will implement this pass now. I assume as long as there does not exist edge between sub-graphs, we can merge enables. Comment on this issue if I miss anything

sgpthomas, 2020-04-02:
This is being addressed in #102.

rachitnigam, 2020-07-10:
Since the rewrite using groups and go/done proposal (#139), the discussion above is stale.

New problems from the groups discussion:
- We somehow need to know that a structural component is shareable. For example, a combinational adder is shareable but an adder that counts the number of times its called is not.
- Perhaps we need a `reset` hole which allows the context to know that any internal state is going to be reset.

rachitnigam, 2021-01-24:
Too stale to be actionable. We should attempt to revive this in a new issue.

closed 2021-01-24 by rachitnigam