calyxir/calyx #76: simple loop unrolling

sgpthomas, 2020-03-19:
A simple pass that finds while loops and unrolls them by some specified factor. This would involve duplicating both the control and the structure by the factor and generating the muxing logic to mux memory accesses. We can restrict the unrolling factors in some way (probably in the same way that dahlia does) so that the muxing logic is not too complicated to generate. Additionally, because Futil has generalized while loops, we don't necessarily have something like a loop incrementation variable. For the purposes of this pass, we should assume something about the shape of the loop. Maybe that it has a single incrementation variable that uses `std_incrementor`.

rachitnigam, 2020-07-10:
@sgpthomas @sampsyo Is this still something we want to do in Futil? From the recent discussions, we were talking about doing this in the Dahlia frontend.

sampsyo, 2020-07-11:
Yeah, I think for now at least this is out of scope and best done in the frontend. A proper unrolling pass, of course, would need to know something about the iteration pattern—something that's not currently true with our all-`while` loop.

It's not something we should do soon, but something we could consider is a "parallel for" control construct that includes a notion of the iterator. (Frontends would pair this with `while` to get an unrolled loop.) Not sure if that would ever actually be useful.

rachitnigam, 2020-07-12:
Sounds good! Closing for now.

closed 2020-07-12 by rachitnigam