calyxir/calyx #89: [Disucssion] Are parallel enables always equivalent to a single enable?

sgpthomas, 2020-03-24:
This is fundamentally a question about the semantics of the `enable` control statement. Consider the following circuit:

[image]

What should the semantics of this control be:
```lisp
(par (enable B + 2 x)
     (enable A * 2 y))
```
So far we've discussed two interpretations of `enable`
  1) `enable` only allows communicate between the sub graph induced by the specified vertices.
  2) `enable` simply lets the sub component "run" (by enabling a valid signal or something like this). Importantly, this would let separate enable statements communicate with each other.

For the above circuit, semantics 1 would put the result of `B + 2` into `x` and `A * 2` into `y`. This program implies implicit muxing of `A` and `B` for the right input of `+`. 

For semantics 2, this program would cause an error or be undefined because `A` and `B` are driving the same wire at the same time. 

I think that semantics 2 has the benefit of being a little more explicit and being closer to hardware. I think semantics 1 might be overall better though. First, it will be easier to compile higher level languages to Futil with semantics 1 because you don't have to worry about muxing at all. Second, if parallel composition of `enable`s is equivalent to a single `enable` in a particular case, then we know that the circuit has the proper muxing in place. In other words, I think that these semantics provide more flexibility to the front end while still being close enough to hardware.

kwf37, 2020-03-24:
So in semantics 1 is `par` kind of a construct for frontends that tells the compiler, "Please give me magic muxing between any components that need it?"

rachitnigam, 2020-03-26:
@sgpthomas reminder to summarize the solution to this.

sgpthomas, 2020-03-26:
We resolved this in favour of semantics 1. To summarize, the `enable` statement only allows communication between nodes specified in the statement. If there are edges coming or leaving the subgraph specified, we pretend these wires don't exist.

This makes it easier to write frontends for Futil because frontends don't have to worry about generating muxing for variables and components. The reasoning about inserting muxes can be done completely in Futil.

These semantics increases the set of valid Futil programs at the cost of reducing the amount of Futil that directory corresponds to hardware. However, because Futil can still represent hardware constructs, this actually increases what reasoning we can do *in* Futil.

We also decided to enforce the restriction that the children of `par` should only be allowed to `enable` disjoint components.
```
(par (enable A B) (enable C D)
```
is legal, but
```
(par (enable A B) (enable B C))
```
is not because `B` is enabled in both children.

This restriction is because `par` performs absolutely no synchronization. It is undefined what will happen if you try to read, write, or use a component in parallel (because you don't know if something else is using the same component). The restriction simply rules out such undefined behavior.

closed 2020-03-26 by sgpthomas