calyxir/calyx #95: Specify implementations for library files

sgpthomas, 2020-03-26:
When generating Verilog or running the interpreter, we need a way to "link" in a concrete implementation for each primitive. There are two main tasks to complete this

- [ ] Decide on some syntax for specifying the file to link. One idea is to augment the current library syntax to take in a list of pairs specifying where the implementation lives for a particular backend:
```racket
(define/library ((verilog ../sim/lib/std.v)
                        (interpreter primitives.so))
  (define/prim std_reg (p_width) ...))
```
A simpler thing to do would just be to pass in the implementation to the command line. I personally don't like this as much and starts making the command line interface rather burdensome. I could be convinced though.
- [ ] Passing this information in a nice way to the backend. By this I basically mean that each backend should just get something like a resolved `PathBuf` pointing to the file specified and not have to worry about details like following relative paths. This is probably as simple as augmenting the `backend::Backend::emit` function to be able to use this information and rewriting `backend::Backend::run` to do most of the heavy lifting.

kwf37, 2020-03-26:
I think we also briefly discussed the idea of implementing primitive components directly in Rust- would that make this task easier?

sgpthomas, 2020-03-26:
I think this is not directly related to the interpreter. Regardless of what the interpreter does, we will need something like this for the Verilog backend (and probably most other backends). I think it would be neat at some point to implement the Futil primitives as a shared library in rust and dynamically load this. This would just separate the interpreter from the primitive definitions a little more which would be nice. However, I think this is probably low priority and we should focus on getting things working with the interpreter first.

sampsyo, 2020-03-27:
Slightly more expansively, I think a combination might be nice:

- Like option 1, the Futil interface declarations coexist with the Verilog implementations.
- Like in option 2, the implementations can be backend-specific.

One way to do that would be to have libraries like this be "fat." (I'm borrowing the term from "fat binaries," which contain machine code for multiple architectures—like PowerPC and x86 during macOS's transition to Intel.) That is, a library consists of an interface declaration and multiple backend-specific implementations—most of which might just be different Verilog files.

These "fat" libraries might _also_ contain a behavioral implementation for the interpreter.

rachitnigam, 2020-04-17:
Fixed by #121

closed 2020-04-17 by rachitnigam