calyxir/calyx #97: Should `valid` ports be explicit?

sgpthomas, 2020-03-26:
For some background, this issue came up when I was implementing the Verilog backend in #94. To compile away `if` statements, we generate wires from the condition port to the valid inputs of the components specified in the true branch and the inverse of the condition port to the valid inputs of the false branch components. This creates a conflict when generating the sequential control FSM in Verilog because for that I take the schedule specified in the Futil and generate wires from the FSM to the valid inputs of all the components. Because I'm generating connection to `valid` in two different places, I end up with verilog that looks like:
```verilog
  std_const #(32, 20) const3 (
      .valid(gt0$out),
      .out(const3$out),
      .valid(const3$valid)   // this conflicts with the earlier .valid connection
  );
```

This leads us to the question, should we ever allow connecting wires to the `valid` port in Futil? Doing so implies that Futil programs can get around the enable semantics by enabling parts of the structure themselves by setting `valid` wires to high. This clashes with the enable semantics where we only allow components specified in the statement to communicate.

I'm thinking that these `valid` ports should only ever be usable by the backend, and that passes should have some other mechanism to disable components from running.

rachitnigam, 2020-03-26:
Just to make sure I understand what’s going on:
- there is a sequential FSM for the true branch that adds valid signals.
- there is the “if” FSM that enables all the components if the condition was true.

The “obvious” answer is that the valid signal should be ANDed so that you get both the signals. The thing that confuses me more is that I would have expected this AND transformation to already happen in Futil passes and not be a problem here. Are these FSMs never explicitly represented in the Futil program?

sgpthomas, 2020-03-26:
ah sorry, I maybe shouldn't have used the term FSM. I'm talking about the implementation of the Futil schedule in the backend. We are assuming that Futil is in the form `(seq (enable A B) (enable B C) ...)`. One solution could be to make this schedule in Verilog an explicit component in Futil. But I'm not sure how much I like having "special" components. And this component would have to be "special" because it's interface (inputs and outputs) depends on the Futil control (the outputs would be all the components used in the control). The backend would then have to find these special components and implement them.

Furthermore, I think there is a semantics problem. If these `valid` ports are explicit, then you could translate
```lisp
(enable A B)
```
to
```lisp
; new structure
[-> (@ A out) (@ B valid)]
; modified control
(enable A)
```
and expect the semantics to be the same. But if we have the semantics where `enable` only allows communication between the components it specifies, then this translation is not valid. If we say that this translation is wrong and the correct translation is to add the structure, but not modify the control (so it stays `(enable A B)`) then the backend doesn't know if it should generate the `valid` signal for `B` or not. This example is maybe less contrived if the behaviour you want is "if `(@ A out)` is high, then invert the `(@ B valid)` signal"

(let me know if this last part is not clear)

tissue3, 2020-03-26:
Maybe we don't need `valid` unless the backend is FSM.
But once we specified backend as FSM, it's fine to remove some structure, e.g `const` out of `enable` semantics, i.e. the lower level Futil program can have backend specified passes. 

Anything connecting to FSM `valid` should correspond to an explicit port. In the case of memory component like register, `valid` from FSM control should connect to `write_en`.

sgpthomas, 2020-03-26:
We resolved this by deciding to move the 'sequential FSM' from being generated in the backend to being generated by passes in Futil. We decided that it is fine for the `valid` port to be used directly. As Zhijiang pointed out, this opens up the ability to potentially optimize away some unnecessary wires depending on the backend.

closed 2020-03-26 by sgpthomas