calyxir/calyx #111: Mux Insertion

sgpthomas, 2020-04-02:
This pass should detect instances of two wires going into the same port and insert muxing logic. After this pass runs, we should be in a form of Futil where there is at most a single wire going into each port. 

We have to think more about how exactly to generate the signal that chooses between the inputs of the mux.

sgpthomas, 2020-04-02:
I'm realizing that this may not be possible to do in general. We probably should have a series of passes that do this in specialized cases. I'm going to turn this issue into a tracker for such specialized passes.

sgpthomas, 2020-04-02:
- [ ] Par Remove by Mux Insertion #113

rachitnigam, 2020-07-09:
Futil has been redesigned to use go/done holes and groups in #139.

closed 2020-07-09 by rachitnigam