calyxir/calyx #113: Par Removal by Mux Insertion

sgpthomas, 2020-04-02:
This pass should remove parallel enables where there are edges between the enable sub-graphs.
For example, consider the following structure:
```
+-+   +-+
|A+--->C|
+++   +++
 |     |
+v+   +v+
|B|   |D|
+-+   +-+
```
In this case `(par (enable A B) (enable C D)` is not equivalent to `(enable A B C D)` because of the edge `A -> C`. This pass should generate a mux that transforms the structure into this:
```
 +-+    +-+    +-+
 |A+---->m+---->C|
 +-+    +-+    +-+
  |             |
 +v+           +v+
 |B|           |D|
 +-+           +-+
```
and modifies the control to be
`(enable A B C D)`
This pass will also have to find when the old edge `A -> B` was used and change the control to
`(enable A m B)`.

tissue3, 2020-04-09:
That's a tri-state buffer. MUX should have >= two inputs.

sgpthomas, 2020-04-09:
I'm using the term mux pretty losely. I just mean some hardware to choose between some number of input signals. Also, my understanding is that tri-state buffers don't exist on FPGAs. Is that correct?

tissue3, 2020-04-09:
I think so. At implementation, they are all converted to lookup table.

tissue3, 2020-04-09:
For this issue, I feel we might need some way to specify the ports that are enabled, not the component. A component can have multiple input and output ports. What does it mean by `enable` a component? All the ports are enabled at the same time. That will cause some function/operation impossible to be represented. `enable` port rather than component may also help the FIFO representation.

sgpthomas, 2020-04-09:
Can you be more explicit about what would be impossible to be represented? I'm a little worried about single port enabled because the semantics become trickier. What does it mean to enable just one port of a component? Presumably the entire component still needs to be run? This means that 'single port' enable implies some kind of implicit hardware to disable the outputs of wires sometimes. I think it may just be better to generate disabling explicitly? I'm not sure though.

tissue3, 2020-04-09:
Port might be the wrong word. Edges/wires might be better?
The example you give is a good one. Also, registers have both read and write ports. What if we just want to read from the register, not writing to it? If we add a tri-state buffer, the register got some `xxxx` signal, but it's still signal! And signals will be updated to the register. I feel tri-state buffer is not enough to represent the graph you draw and registers.

tissue3, 2020-04-09:
I will implement this.

tissue3, 2020-04-13:
Current remove_par pass can change 
```
(define/namespace prog
  (define/component main () ()
    ([new-std x0 (std_reg 4 0)]
     [-> (@ const0 out) (@ x0 in)]
     [new-std const0 (std_const 4 0)]
     [new-std y0 (std_reg 4 0)]
     [-> (@ const1 out) (@ y0 in)]
     [new-std const1 (std_const 4 10)]
     [-> (@ const0 out) (@ y0 in)]
     ;[-> (@ const1 out) (@ x0 in)]
     )
    (seq
     (enable y0 const0)
     (par
      (enable x0 const0)
      (enable y0 const1)))))
```
to
```
(define/component main
  ((port valid 1))
  ((port ready 1))
  ([new-std const0 (std_const 4 0)]
   [new-std const0_y0_id (std_id 4)]
   [new-std const1 (std_const 4 10)]
   [new-std x0 (std_reg 4 0)]
   [new-std y0 (std_reg 4 0)]
   [-> (@ const0 out) (@ const0_y0_id in)]
   [-> (@ const0 out) (@ x0 in)]
   [-> (@ const0_y0_id out) (@ y0 in)]
   [-> (@ const1 out) (@ y0 in)])
  (seq
   (enable y0 const0)
   (par
    (enable x0 const0 y0 const1))))
```
There is one more step we need to perform: changing 
```
   (enable y0 const0)
```
to
```
   (enable const_y0_id y0 const0)
```

Is there anyway more elegant way than adding another pass, i.e. the second part of pass remove_par, so that the visitor goes through enables and check if id is presented?

Or, is `const_y0_id` is already sufficient by its name to illustrate there is an out edge from `const` to `y0`?

tissue3, 2020-04-13:
Because I couldn’t come up with a way that untill `par`  is encountered, `enable` knows it needs to add `id` ahead of time, like in this example
```
  (seq
   (enable y0 const0)
   (par
    (enable x0 const0 y0 const1))))
```
`enable y0 const0` cannot know in `par`, `y0` and `const0` requires separation, so that `id` is inserted between them.

tissue3, 2020-04-13:
Even when we need another pass to finish changing `enable y0 const0`, it would ideal to get the information in the pass, like a hashmap recording modified components. I am not sure how to do it in futil structure.

sampsyo, 2020-04-13:
That's a good point, @tissue3, and something I've been thinking about as well: the "analysis results" that come from detecting conflicts for this pass seem like they should be useful outside of the pass. Generalizing slightly, it would be really nice to have some way for one pass to create intermediate results that multiple other passes can consume to decide whether an optimization is legal, whether it's "in the IR" or on the side. LLVM itself has this for [alias analysis](https://llvm.org/docs/AliasAnalysis.html), for example—and I suspect we will want something very much like conservative alias analysis once we have "loans" (#48).

Of course, on-the-side data structures can be bad too: in the limit, for example, you can imagine that we'd make HLS scheduling results into a non-IR data structure, and that would be bad—we want substantial changes to the program's level of abstraction like that the captured in the Futil IR itself. So if we do this we'll need to be careful to make a hard distinction between analysis results (which are just "caches" of information deduced _about_ a program) from optimization/transformation decisions (which directly dictate what the resulting program will look like).

rachitnigam, 2020-07-09:
Futil has been redesigned to use go/done holes and groups in #139.

closed 2020-07-09 by rachitnigam