calyxir/calyx #116: seq muxing

sgpthomas, 2020-04-06:
We need to resolve multiple wires going into the same port in different logical time steps.
Consider this simple Dahlia program
```C
let x: bit<32> = 1;
---
x := 5;
```
We are simply updating the value of a register. This generates the following Futil code:
```lisp
(define/namespace prog
  (define/component main () ()
    ([new-std x0 (std_reg 32)]
     [-> (@ const0 out) (@ x0 in)]
     [new-std const0 (std_const 32 1)]
     [new-std const1 (std_const 32 5)]
     [-> (@ const1 out) (@ x0 in)])
    (seq
     (enable x0 const0)
     (enable x0 const1))))
```
We need to insert a mux that feeds into `(@ x0 in)` that selects between `const0` and `const1`.

How I'm imagining this will work atm is inserting a mux into the structure, rewiring the inputs, to `(@ x0 in)`, and figuring out some way to generate the selection logic for the mux. The selection logic seems like the trickiest bit.

rachitnigam, 2020-04-13:
Is implementing this going to be similar to #113?

sgpthomas, 2020-04-13:
Yeah, I think it'll be pretty similar. In addressing this question, I came up with an interesting conceptualization:

[image]

The bottom picture is in some sense, the minimum amount of control necessary. We absolutely need some kind of muxing and we need to enable the `write-en` port. (the s0 -> s1 thing is the seq fsm which you need to enforce ordering). 

The picture on the right is a sketch of the Futil implementation. Basically, the `id` components pass through data, and act as a way to add a control signal to a wire. You can then use the control signals to generate the input to the mux and the write-en signal. The control transformations would look something like:
```
(seq (enable x 0) (enable x 1)) 
-> (seq (enable x id_0 0) (enable x id_1 1))
-> (enable x id_0 0 id_1 1 mux s0 s1)
```

sampsyo, 2020-04-14:
Cool; this is a neat way of looking at it. To clarify, though, it seems like you're assuming that there is a "cleanup" pass to run afterward that deletes needless `id` units?

sgpthomas, 2020-04-14:
I'm not sure that's necessary? I think we could implement the verilog for this and hope that they compile it away. I don't really understand enough about the semantics of verilog to be certain, but it seems like we would only be specifying wire connections so the verilog would actually just be equivalent to wires.

Im also not sure if this conceptualization always works. Does this also work for the par muxing we were talking about last meeting?

rachitnigam, 2020-07-09:
Futil has been redesigned to use go/done holes and groups in #139.

closed 2020-07-09 by rachitnigam