calyxir/calyx #119: Latency insensitive pass captures port names

rachitnigam, 2020-04-12:
If there ports named `ready` or `valid` already present in the component, the pass will add new ports with the same names causing a conflict.

The pass should either rename the existing ports when a conflict occurs or generate new names for the `ready` and `valid` ports.

Minimal example:
```racket
(define/namespace prog
 (define/component comp1 ((port valid 1)) ((port ready 1))
  ()
  (seq)))
```

which generates:
```
(define/component comp1
  ((port valid 1) (port valid 1))
  ((port ready 1) (port ready 1))
  ()
  (seq))
```

sgpthomas, 2020-04-13:
I've fixed this in `mvp_verilog`

sgpthomas, 2020-04-13:
https://github.com/cucapra/futil/pull/94/files#diff-f602af88b28120b3485ab34caeb434e9R19

rachitnigam, 2020-04-13:
I don't think that's the right fix. You've given special semantics to named port. What happens if that port is being used for something else?

The proposed solutions above treat the existing wires as if they already have an internal semantics for the program and try to preserve it.

sgpthomas, 2020-04-13:
Yeah you’re right. It would be better to just error out for now then. We need to think more about how to add semantically meaningful ports / wires

PR #94 (2020-04-17): [WIP] Mvp verilog
This is a work in progress PR for implementing the minimum things to get Verilator running on a subset of Futil programs. The subset is the Futil programs that only use `seq`, `par`, `enable`, and `if`. In particular this means:
 - [x] cleaning up Verilator tool chain and writing documentation for how to install this and get it running, add instructions for using Verilator backend
 - [x] implementing pass to wire up the toplevel clock to all components that need it
 - [x] generating the Verilog control FSM to implement the schedule
 - [x] hooking up the signals generated from [... cut]
closed 2020-04-17 by rachitnigam