calyxir/calyx #126: fix remove if

tissue3, 2020-04-14:
Remove if pass is buggy. It changes 
```
(define/namespace prog
  (define/component main () ()
    (; structure
     [new-std a0 (std_reg 32 0)]
     [new-std const0 (std_const 32 0)]
     [new-std b0 (std_reg 32 0)]
     [new-std const1 (std_const 32 1)]
     [new-std gt0 (std_gt 32)]
     [new-std const2 (std_const 32 1)]
     [new-std y0 (std_reg 32 0)]
     [new-std const3 (std_const 32 2)]
     [new-std z0 (std_reg 32 0)]
     [new-std const4 (std_const 32 4)]
     [-> (@ const0 out) (@ a0 in)]
     [-> (@ const1 out) (@ b0 in)]
     [-> (@ a0 out) (@ gt0 left)]
     [-> (@ const2 out) (@ gt0 right)]
     [-> (@ const3 out) (@ y0 in)]
     [-> (@ const4 out) (@ z0 in)])
    (seq
     (par
      (enable a0 const0)
      (enable b0 const1))
     (enable gt0 b0 const2)
     (if (@ gt0 out) (gt0 a0 const2)
         (enable y0 const3)
         (enable z0 const4)))))
```
to
```
(define/component main
  ((port valid 1))
  ((port ready 1))
  ([new-std a0 (std_reg 32 0)]
   [new-std b0 (std_reg 32 0)]
   [new-std const0 (std_const 32 0)]
   [new-std const1 (std_const 32 1)]
   [new-std const2 (std_const 32 1)]
   [new-std const3 (std_const 32 2)]
   [new-std const4 (std_const 32 4)]
   [new-std const4_not (std_not 1)]
   [new-std gt0 (std_gt 32)]
   [new-std y0 (std_reg 32 0)]
   [new-std z0 (std_reg 32 0)]
   [new-std z0_not (std_not 1)]
   [-> (@ a0 out) (@ gt0 left)]
   [-> (@ const0 out) (@ a0 in)]
   [-> (@ const1 out) (@ b0 in)]
   [-> (@ const2 out) (@ gt0 right)]
   [-> (@ const3 out) (@ y0 in)]
   [-> (@ const4 out) (@ z0 in)]
   [-> (@ const4_not out) (@ const4 valid)]
   [-> (@ gt0 out) (@ const3 valid)]
   [-> (@ gt0 out) (@ const4_not in)]
   [-> (@ gt0 out) (@ y0 valid)]
   [-> (@ gt0 out) (@ z0_not in)]
   [-> (@ z0_not out) (@ z0 valid)])
  (seq
   (par
    (enable a0 const0)
    (enable b0 const1))
   (enable gt0 b0 const2)
   (seq
    (enable gt0 a0 const2)
    (enable y0 const3 z0 const4)))) ;   <------ Missing components
```
The RemoveIf pass does not make last two statement together because there is a chance that `a0` is also part of last `enable` statement.  

1. The first issue is to define what it means when we `enable` a component where the `valid` port is already occupied, like `z0` and `y0` in this case. 
2. If `valid` port of a component is occupied by other component, say `z0_not`, shall we remove `z0` out of the last `enable` statement?
3. Because `gt0` is a comparator that cannot store value, we need to add a register to store the output of `gt0` and use the output to `enable` a branch. 
This is not necessary if we can merge all into one line, e.g. `(enable gt0 a0 const2 y0 const3 z0 const4)` in this example. 
4. What if `if ` statement triggers multiple lines? E.g. `if(seq(enable A B)(enable AC)) ()`. Shall we still merge two branchs of `if`? It makes execution time at least the slowest branch.

sgpthomas, 2020-04-14:
3 seems relatively simple to solve. We can just stick in a register, or maybe just merge those two enable statements. 1 and 4 seem much harder and I think are related to the issue of composable passes: #127.

rachitnigam, 2020-07-09:
Futil has been redesigned to use go/done holes and groups in #139.

closed 2020-07-09 by rachitnigam