calyxir/calyx #133: TVM frontend for Futil

sgpthomas, 2020-05-26:
We want to write a TVM frontend for Futil so that we can show how Futil makes it easy to write FPGA compilers for widely used languages. I don't know much about the internals of TVM or how to write a backend for TVM so I can't break up this task into smaller chunks yet. 

 - [ ] figure out what we need for a TVM backend: [docs](https://docs.tvm.ai/dev/relay_bring_your_own_codegen.html)

ViviYe, 2020-07-15:
updates:
already set up a simple structure to traverse relay AST and construct a dummy Futil AST accordingly
TODO:
* implement a simple variable declaration (just make it print as a variable binding)
* start constructing a graph from the relay AST (I might need to look into Dhalia backend for how things should be structured)
* write out several small examples (eg: simple variable assignment, and identity function)and make the compiler compile these small examples

rachitnigam, 2020-10-29:
This is out of date. @cgyurgyik has written up some thoughts in #267.

closed 2020-10-29 by rachitnigam