calyxir/calyx #140: Optimizations

sgpthomas, 2020-07-09:
- [ ] Don't generate priority logic for muxes: https://sutherland-hdl.com/papers/2005-SNUG-paper_SystemVerilog_unique_and_priority.pdf
 - [ ] Redundancy elimination
 - [ ] Scheduling
 - [ ] Resource Sharing
 - [ ] Pipelining
 - [ ] Bitwidth analysis
 - [ ] Control pipelining

rachitnigam, 2020-07-10:
What's "redundancy elimination"?

sampsyo, 2020-07-11:
That was the notion we discussed briefly about a "final-stage" cleanup to eliminate parts of guards like `& 1` that are artifacts of earlier passes.

rachitnigam, 2020-08-24:
Created separate issues for resource sharing (#197) and bitwidth analysis (#198). The others are either not optimizations (pipelining) or unclear how to implement them (control pipelining).

rachitnigam, 2020-12-27:
Added one-hot encoding optimization for generated FSMs.

rachitnigam, 2024-07-22:
This is generic and in-actionable enough that I will close this

closed 2024-07-22 by rachitnigam