calyxir/calyx #144: Pass to create FSM state from all guards in an inlined module

rachitnigam, 2020-07-12:
Post inlining, our generated code looks (equivalent to):
```verilog
if (g1 | g2) x1 = y1;
else if (g3) x1 = y2;
else x1 = '0;

if (g1) x2 = z1;
else if (g2) x2 = z2;
else if (g3) x2 = z3;
else x2 = '0
```

We can create a pass that does two things:
1. Name every guard expression so that the conditions are more readable. This is a purely readability pass.
2. Walk over every guard condition and coalesce the program into:
```verilog
if g1 begin
  x = y1;
  x2 = z1;
end else if begin
  ...
end
```
Next, we can turn `g1`, `g2`, `g3` into a one-hot vector and use that to implement a module-level FSM. (Assuming that only one guard condition is ever active).

sampsyo, 2020-07-12:
Sounds cool. One thing I'm a little unclear on here, however, is how that latter program will be represented in Futil—won't we still have a list of assignments with separate guards, even if they're named?

rachitnigam, 2020-07-12:
Yeah, so the odd thing about the current setup we have is that all of these optimizations (including #143) happen in the code-gen phase. One way to explicitly represent (1) is transforming:
```
x = long_guard_cond ? y
```
into
```
_cond = 1 ? long_guard_cond;
x = _cond ? y;
```
which still accomplishes the problem of naming the guard conditions.

sampsyo, 2020-07-13:
Got it, that makes sense! Although I think given the current language, the first line of the second block would have to be:

```
_cond = long_guard_cond ? 1;
```

because the syntax for these assignments is `<id> = <cond> ? <id>`, i.e., conditions are allowed to be complex expressions but values are only allowed to refer to other signals by name.

One thing we have talked about a couple of times is introducing a special category guard variables, so you could do this:

```
_cond = long_guard_cond;
```

*just* for the purpose of giving reusable names to big guard expressions. `_cond` here would have a type that prevents it from getting mixed up in actual value signals, I assume, and would limit it to use only in guards.

rachitnigam, 2020-07-13:
I'm realizing that (1) just undoes the work that the inliner is doing. The inliner syntactically inlines all of these conditions. Instead of doing that, it can compute those values on a wire and then use that value.

sampsyo, 2020-07-13:
Right, but doing that (computing the condition values on a wire) would require some nuance: seems like there are three options.

- Using the `cond ? 1` trick.
- Introduce special wire-like values that can hold condition expressions.
- Convert the logic expression to structure (probably don't want to do this one).

Basically, condition logic and regular structural logic currently exist in two separate realms and ne'er the twain shall meet. Which has its advantages: only conditions require logic synthesis, for example, and structure is basically a netlist.

rachitnigam, 2020-10-06:
I'm no longer sure if this is a good idea. #143 introduced the bug in #235 which means my understanding of if-else blocks might be wrong.

closed 2020-10-06 by rachitnigam