calyxir/calyx #145: Evaluation

rachitnigam, 2020-07-15:
Tracking things for evaluation:

Note on "evaluation": We need to figure out what axes of evaluation we want. Choices (in order of hardness) simulation cycles, timing, area, energy, FPGA runtime.

## Bare minimum
**Benchmark suites** (subsets of): Polybench, MachSuite, [Dynamatic][] benchmarks (if any)
- [ ] Evaluate *non-pipelined* static timing against Vivado HLS.
- [ ] Evaluate *non-pipelined* dynamic programs against dynamatic. This is not apples to apples because Dynamatic is pipelined by default.

## Good to have
- [ ] Evaluate *pipelined* static timing against Vivado HLS.
- [ ] Evaluate *pipelined* dynamic programs against dynamatic.

[dynamatic]: https://dynamatic.epfl.ch/

sampsyo, 2020-07-15:
Including a link to [the Dynamatic source code](https://github.com/lana555/dynamatic) for completion. It would be great to check relatively soon whether it's easy/possible to compile and run.

rachitnigam, 2020-07-17:
@sgpthomas As we get the various benchmarks working, we should make them integration tests and run simulation on them.

closed 2020-08-24 by rachitnigam