calyxir/calyx #153: Static timing implementation proposal

rachitnigam, 2020-07-20:
Add statically timed groups to Futil. Note that this proposal does attempt to implement [SDC scheduling](https://www.csl.cornell.edu/~zhiruz/pdfs/sdc-dac2006.pdf) (**TODO**: explain why and when we can add SDC scheduling).

### Syntax extension

Add the following syntax to groups:
```
group foo[k1=n1, k2=n2] { ... }
```
where `k` is a double-quoted strings and `n` is a decimal number.

sampsyo, 2020-07-21:
Awesome! I like this direction a lot.

**Syntax:** Seems logical overall! I see your point about all the brackets getting somewhat cluttered. I think we should forge ahead with this design because (a) concrete syntax isn't all that important and (b) let's go for the smallest/easiest change first, as usual. Maybe in the future we can consider a grand reset on all the bracket syntax and moving to a "bag of keywords" style, like this:

	primitive std_reg {
		param p_width;
		input in: p_width;
		output out: p_width;
		output done: 1;
		attr static = 3;
	}

…but no need to do anything like that now.

**Real-time latency:** I agree that `"static"=0` should be allowed for combinational components. AFAICT these do not need any special handling and can be exploited in exactly the same way as other components with positive cycle latency (for "done elimination," etc.).

*…until* we try to do "operator chaining," i.e., trying to pack multiple computational operations into a single cycle. In my view, this is best left to a future proposal because it requires a different kind of information: real-time latency (i.e., nanoseconds) instead of cycle latency. We can cross that bridge later, probably by adding another attribute that optionally gives the nanoseconds for any "chainable" primitive (which must have `"static"=0`).

**Statically timed FSMs:** Generally looks good, and I agree with v1 "giving up" entirely on any `seq` that is not 100% uniformly statically timed (falling back to the current dynamically timed FSM generation). The decision to relax this restriction can be data-driven.

**Representing components that are used in a static way:** Q3 and Q4 are quite related: basically, they ask "how do we represent components that may be used *either* in a statically timed *or* a dynamically timed way?" When *using* the component (Q3), is it OK to just ignore the `done` signal and otherwise use exactly the same code as when we were using the dynamically timed interface to the same component?

I like exactly what's in this proposal, I think. But to expand on the reasons:

- For the interface, let's try to think of the semantics of the `done` signal "calling convention" as follows. Basically, you (the caller) need some way to assure that you only read the outputs after they are ready. One way to do that is to ensure that you don't use the outputs until the `done` signal goes high. That way is nice because it's universal and it's easy to see statically in the code when you've done it correctly. But it's not the _only_ way to know that it's safe to read the outputs of a component, i.e., to know that it's safe to run the next stage in a `seq`. A different way to do that would be to count cycles and wait long enough for the component to be done.

  Put differently, the invariant that FuTIL wants to enforce on programs is not that parent components read `done` for all their child components. That's just a means to an end: waiting on `done` is just a _sufficient but not necessary_ way to ensure that you don't try to consume values before they are ready. 

- For the definitions, it's a good point that some components might want to have different implementations for when they are used in a dynamically timed way vs. a statically timed way—especially if they are implemented in black-box Verilog! Not all components will want this, but I'm comfortable with having a rewriting pass that knows about a few baked-in primitives like `std_reg` that want this treatment and substitute one for the other when used in a statically timed context.

**SDC scheduling:** IMO we should leave this for later, and for now just make sure we are not painting ourselves into a corner that would somehow prevent implementation of SDC scheduling. In other words, this statically-timed-FSM-generation thing can be thought of a very basic form of static scheduling that does not attempt to do any parallelization. Ideally, we will build up abstractions now that will enable SDC (or other sophisticated scheduling passes) to be "dropped in" modularly.

sgpthomas, 2020-07-27:
**Q1**: I agree that for now we should leave the syntax the way it is. At some point we can evaluate a change that makes it use fewer different pairs of brackets.
**Q2**: I agree with Adrian that `"static"=0` is right here. I think that the way to think about this is that `"static"` is referring to the number of clock cycles that the component / group will take. It does not talk about real latency in ns at all. For that we will need some other annotation like `"realtime"` that provides nanoseconds. With that, we would be getting into real backend (the actual target FGPA or other hardware) specific optimizations and we should think about the best way to modularize this. I think the existing `.lib` file infrastructure should be enough but it's worth another think about.
**Q3**: I think yes is the answer here and I agree with Adrian on the explanation for why.
**Q4**: I think in the short term we shouldn't worry about this and in the long term it might make sense to have go/done holes be more first class for primitives. Right now, primitives can just declare ports that have the same name as a hole and the rest of the compiler treats it as a hole. This doesn't seem ideal. I think the right solution may revolve around how the existence of holes interacts with the `static`/other annotations. What I mean here is that when a primitive is compiled using the `static` pass, it's done hole will not be used. When it's compiled with the normal `dynamic` pass, it will definitely be used. To communicate this, we could provide different primitive implementations based on the annotation used:
```C
primitive std_reg[width](in: width, write_en: 1) -> (out: width) {
  verilog<"static"=1> {
    verilog implementation without done hole
  }

  verilog {
    fallback verilog implementation with done hole
  }
}
```
**NOTE**: This is very similar in function to Rachit's proposal, it's just maybe a little more structured.

**Q5**: Again, I agree with Adrian here. We should make sure that we don't do anything that makes this impossible, but we don't have the time to work on this right now.

PR #155 (2020-07-28): Static timing implementation
Fixes #153.

Implements the `StaticTiming` pass that compiles groups using static FSMs when
possible and calculates the static of a group.

This PR implements
- The static timing proposal
  - Adds attribute syntax `group g<"k"=n, ...> { ... }`
  - Implements a pass that uses the `static` attribute to generate groups that can use self incrementing counters to enable and disable `go`/`done` holes.
- A pass manager
  - Adds the `register_pass!` and `register_alias!` macros.
  - Adds the `-d` flag to the CLI to disable a pass or an alias.
- New macros for writing struc [... cut]
closed 2020-07-28 by rachitnigam