calyxir/calyx #159: Top down static timing pass

rachitnigam, 2020-07-27:
The pass implemented #155 is a bottom-up pass. It compiles the inner-most groups and proceeds upwards. This results in one "counter" fsm being generated for each nested control statement which is pretty wasteful. A different strategy is pre-processing the FuTIL program to decide which subtrees can be statically schedules and then generating one schedule for the entire control subtree and use one "counter" FSM.

rachitnigam, 2020-11-25:
Unclear about the usefulness of this pass. Closing for now.

rachitnigam, 2020-12-20:
I was wrong. I think this pass will be useful.

Consider a control program like (where one take 1 cycle, two takes 2 cycles, etc.):
```
while {
  seq {
    par { one; two }
    three;
  }
}
```

There are three FSM controllers generated for this: 
1. `fsm_par` that counts up to 2 (enabling `one` for 1 cycle and `two` for 2 cycles)
2. `fsm_seq` that counts up to 5 (enabling the `par` group for the first 2 cycles and enabling `three` for 3 cycles)
3. `fsm_while` that repeatedly counts up to 5 and enables `seq` group.

Here is the problem: all of these `fsm` are counting up at the same time! Worse still, the outermost FSM is already using the largest numbers of bits needed to encode the entire control flow.

If we do a top-down pass, we should just be able to use `fsm_while` to control the whole control program.

sampsyo, 2020-12-20:
Super cool point; this would be fun to evaluate.

rachitnigam, 2021-10-07:
Subsumed by #654

closed 2021-10-07 by rachitnigam