calyxir/calyx #162: Getting real VHLS comparison numbers

rachitnigam, 2020-07-29:
Adrian and I discussed more ideas for getting real resource numbers (in order of effort):
1. Don’t generate the interface pragmas. 
2. Force the Dahlia generated code to not use DSPs using pragma magic (there must be some way to disable DSP instantiation?)
3. Change the memory interface Futil programs use to be the same as the ones used by VHLS. There are two options here:
  - Change all VHLS memories to be local arrays. We can separate out the kernel into a new function and the `main` kernel simply moves things into local arrays and passes them to this new function. Then, we can report the  resource numbers for the new function.
  - Use the specifically instantiated memories that VHLS is using. I *think* these memories have a one cycle read latency which would affect the code generation from the Dahlia backend.

For (3), I would recommend first experimenting with some hand written code instead of committing to changing the Dahlia compiler.

rachitnigam, 2020-07-29:
The following report shows that DSPs are actually inferred for the generated FuTIL program but are never instantiated in the final design: [report.txt](https://github.com/cucapra/futil/files/4995694/report.txt)

The lower effort thing still seems to be (2) where we force the VHLS code to not generate any DSPs.

sampsyo, 2020-07-30:
Yeah, I think it is true that the "path of least resistance" is somehow convincing Vivado HLS to _not_ use DSPs for arithmetic operations (versus trying to make FuTIL actually use DSPs). However, the report above makes me question that conclusion… if there were some secret incantation we need to say just to get those DSPs to actually go all the way through synthesis, maybe that would be even easier?

To quote the Vivado HLS output:

> The table above is a preliminary report that shows the DSPs inferred at the current stage of the synthesis flow. Some DSP may be reimplemented as non DSP primitives later in the synthesis flow. Multiple instantiated DSPs are reported only once.

It hints that the implementation step may remove some DSPs, but it doesn't say way. I'm going to dig into this a little bit more and see if there's an easy answer here…

rachitnigam, 2020-07-30:
Trying some google foo: https://stackoverflow.com/questions/57648144/how-can-i-force-vivado-to-use-dsp-blocks-for-all-arithmetic-operations

sampsyo, 2020-07-30:
Our interface looks like this:

```
input wire clk,
input wire go,
output wire done
```

This comment advances a theory that what we're seeing here is not a failure to bind our `*`s to DSPs but a problem with our interfaces that is leading to DCE. Vivado doesn't see us "interacting with the outside world" enough and therefore assumes that lots of our code is dead, so it can eliminate some memories and arithmetic operations. The evidence is not conclusive but at least worth exploring.

## Background & Forming a Hypothesis

I tried running a FuTIL example through Vivado: I used `linear-algebra-atax.expect` from our benchmarks directory and deleted the unsynthesizable memory-initialization stuff. The Verilog contains four memories, `A`, `x`, `y`, and `tmp`, instantiated as `std_mem_d{1,2}` components. It also has two `std_mult` multiplier components, `mult0` and `mult1`.

Early in the synthesis log, we can see evidence of those components existing. Such as this table that @rachitnigam pointed out, and this interesting but not-too-worrisome warning about how the synthesis is likely to map one of our memory to registers:

```
INFO: [Synth 8-6157] synthesizing module 'std_mem_d2' [/home/als485/vivado_test/atax.sv:3]
	Parameter width bound to: 32 - type: integer 
	Parameter d0_size bound to: 8 - type: integer 
	Parameter d1_size bound to: 8 - type: integer 
	Parameter d0_idx_size bound to: 4 - type: integer 
	Parameter d1_idx_size bound to: 4 - type: integer 
WARNING: [Synth 8-5856] 3D RAM mem_reg  for this pattern/configuration is not supported. This will most likely be implemented in registers
```

But sure enough, we get an eventual report that uses zero DSPs and zero memories of any kind.

Posts around the Internet seem to be suggesting (again, inconclusively) that primitives like DSPs and memories should be inferred automatically—so it's at least suspicious that we are not seeing any of these end up in the final counts.

So here let's form a hypothesis that much of the design is actually being eliminated because it's not "hooked up" to the outside world. That would sort of make sense: we're instantiating memories internal to the `main` SystemVerilog module. The input memories are never written to, but we read from them as if they contain useful data. The output memories are never read from. A sufficiently smart RTL synthesis tool might conclude that they are unnecessary, as are all the bits of hardware along the data path that computes from/to these memories.

Obviously, Vivado HLS does not have this problem. So the next step I was interested in was comparing how Vivado HLS works on a simple example.

## Comparing to HLS 

I tried a real simple Vivado HLS implementation of a dot product:

```
int myfunc(int a[32], int b[32]) {
    int out = 0;
    for (int i = 0; i < 32; ++i) {
        out += a[i] * b[i];
    }
    return out;
}
```

I compiled this to Verilog and just checked it out manually. For such a small design, it's not an overwhelming amount of code. I was interested in two questions: is it generating any kind of extra constraints to say "please use a DSP"? And what does the interface look like, hypothesizing that whatever it's doing permits it to avoid the bad DCE we're seeing?

The first question was sort of a bust. I couldn't find any interesting directives anywhere, DSP-related or no. I looked pretty hard, but it still doesn't mean they don't exist.

But the interface question was more interesting. Here are the Verilog wires that make up the interface for the synthesized dot product:

```
input   ap_clk;
input   ap_rst;
input   ap_start;
output   ap_done;
output   ap_idle;
output   ap_ready;
output  [4:0] a_address0;
output   a_ce0;
input  [31:0] a_q0;
output  [4:0] b_address0;
output   b_ce0;
input  [31:0] b_q0;
output  [31:0] ap_return;
```

[... 12 lines cut ...]
* How do we synthesize these designs? (As @sgpthomas previously noted, it's not obvious how to feed Vivado HLS outputs with all those extra interface wires through Vivado synthesis. If you don't do anything, you will get lots of errors about unconnected ports, which makes sense.)

## How to Synthesize Designs with Interesting Interfaces

So far, the way we've been synthesizing FuTIL code is by explicitly hooking up `go`, `done`, and `clk` to actual, physical pins on the FPGA. It seemed to me like this is very unlikely how VHLS works for synthesis (it would have to guess which pins to hook up each signal to!)—it's probably doing something more "disembodied" where it just synthesizes the logic for the function in isolation, hoping that you'll wire it up to other RTL or whatever to feed it with data.

So how can we match that behavior for FuTIL designs? I can think of two ways:

1. Suggested by @rachitnigam yesterday: we create a Vivado HLS wrapper that invokes the FuTIL code as a black-box RTL function.
2. We try to replicate however Vivado HLS is synthesizing these logic blocks in isolation.

The latter seemed instructive to at least try. I dug into this by looking at the `run_vivadosyn.tcl` that Vivado HLS generates and feeds into Vivado for synthesis. 😭 It seems to use a mode called [out-of-context (OOC)](
https://forums.xilinx.com/t5/Synthesis/How-to-perform-Out-of-Context-Synthesis/td-p/958640) synthesis:

```
set_property -name {STEPS.SYNTH_DESIGN.ARGS.MORE OPTIONS} -value {-mode out_of_context} -objects [get_runs synth_1]
```

And the `.xdc` file that we are using to hook things up to pins instead just contains a clock, nothing else:

```
create_clock -name ap_clk -period 7.000 [get_ports ap_clk]
```

As a next step, I want to figure out how to replicate this setup ourselves for FuTIL RTL. Then maybe we can even feed the FuTIL and VHLS RTL through the same backend synthesis script!

sampsyo, 2020-07-30:
OK, I worked out the whole "out-of-context" thing. It's demonstrated in my little Vivado demo repository:
https://github.com/sampsyo/vivado-example

There are several changes to note, including a simpler `.xdc`, a different "stop point" for the implementation step, and an extra config command. But the upshot is that I was able to actually synthesize Vivado HLS-generated Verilog! (The only differences were that VHLS files are Verilog instead of SystemVerilog; the function name is not `main`; and the clock signal has a different name.) I think we should use this mode for future experiments!

sgpthomas, 2020-07-30:
So, thanks to Luis and [this](https://github.com/vegaluisjose/dsp-example/blob/master/tcl/vivado.tcl) repo, I discovered the `(* use_dsp = "yes" *)` annotation in Verilog that causes us to synthesize DSPs. I'll continue to do some investigation but it looks like this is sufficient and we don't need to do the memory interface stuff.

I think the next question is getting BRAM numbers, which may cause us to need to implement the `ap_memory` interface as Vivado does, but maybe there's a similar annotation for BRAMs.

rachitnigam, 2020-07-30:
Hold on, we need real LUT usage numbers before we move to BRAM. To do that, we need to make changes to the Dahlia front end to generate the same memories as VHLS.

sgpthomas, 2020-07-30:
How were you getting BRAM numbers for Dahlia? All of the numbers that we are getting from vivado_hls for Dahlia programs are 0 for memories of any kind.

rachitnigam, 2020-07-30:
Link? The 2mm file has BRAM counts: https://github.com/cucapra/futil/blob/master/vivado-experiments/results/2020_07_28_11_18/linear-algebra-2mm/kernel_csynth.rpt

Also, unless you’re looking at BRAM for debugging purposes, LUT numbers are more important.

sgpthomas, 2020-07-30:
ah, this is when I turn off the memory interfaces in the Dahlia vivado backend

sampsyo, 2020-07-30:
> I'll continue to do some investigation but it looks like this is sufficient and we don't need to do the memory interface stuff.

Yo, this conclusion seems a little hasty—as I mentioned in my (overlong) post above, there are two reasons to do the interface thing:

> - apples-to-apples comparison with VHLS
> - it _might_ (although it's unproven) avoid unwanted DCE

The first still seems important. And I don't think we have proven yet that DCE is not a problem, right?

sampsyo, 2020-07-30:
> ah, this is when I turn off the memory interfaces in the Dahlia vivado backend: https://github.com/cucapra/dahlia/tree/vivado_no_memory_int

Yes, we added these `m_axi` and `s_axilite` primitive declarations in Dahlia because they are a way to force Vivado HLS to put make interface BRAMs _internal_ to the RTL that it generates, so they're included in overall resource estimates.

Overall, I am not sure that we need to do the same thing (that is, include BRAMs for interface memories) for FuTIL! We could instead mirror the default `ap_memory` behavior of Vivado HLS, which "externalizes" interface memories but still uses locally-declared BRAMs for internal temporaries. This would yield (a) an apples-to-apples comparison, and (b) a realistic design point for generating standalone "IP blocks" that work on memories not under their own control.

rachitnigam, 2020-07-30:
[This thread](https://forums.xilinx.com/t5/Synthesis/Vivado-2013-1-removing-quot-unused-quot-logic-need-to-understand/td-p/331557) also has some DCE debugging guidance. In particular, it looks like we have to stare at the synthesis logs to make sure that it doesn't have messages like:

```
Synth 8-3295 tying undriven pin module_nam:port_name to constant 0
```

sgpthomas, 2020-07-30:
ok, I got some numbers that looked real. As expected, you have to hook up the inputs and the outputs to the interface otherwise it is removed. For dot-product I manually edited the verilog to have A, B and v all declared as part of the interface. This made things work.

I think we'll need something like the verilog `input` and `output` keywords in the cell section. Internally in the compiler, we'll treat these as just normal components, but then we'll have a pass that removes the cell declaration and plops them in the interface. With this, I don't think the backend should have to change much or at all.

sampsyo, 2020-07-30:
These are helpful finds, @rachitnigam! One thing that comes to mind is that the "undriven pin" thing might not show up if the reason for DCE is a blank _memory_ rather than an unassigned `logic` variable.

And that sounds good, @sgpthomas—let's talk about the design in a few minutes. :smiley:

rachitnigam, 2020-08-02:
@sgpthomas close this if you think we're done.

sgpthomas, 2020-08-02:
In the name of keeping a record, the hypothesis that Vivado was removing the bulk of our design was correct. We solved this for memories by:
 - introducing `std_mem_d{n}_ext` primitives that act like normal memories but then are externalized into the module interface. We run synthesis in out-of-context mode and this causes Vivado to not remove the design.
 - make Dahlia generate memories using the `ap_memory` interface which similarly externalizes the memories. This allows for a fair comparison.

We also now gather latency numbers from Vivado HLS. For this we just had to make sure that `while` loops have accurate trip counts. Polybench has a number of 'triangle' loops of the form:
```python
for i in range(8):
    j = 0
    while (j < i):
        j += 1
```
This means that VHLS generates different numbers for `avg`, `min`, and `max` latencies. I think that `avg` is the most fair comparison against Futil because for triangle loops, `avg * loop body latency` will be close to the actual latency of the loop. We might also draw `min` latency lines on our graph to benefit the competition.

closed 2020-08-02 by sgpthomas