calyxir/calyx #176: Tests change too often

rachitnigam, 2020-08-04:
Most of our PRs currently come with thousands of changes and often change the futil lowered for verilog code we generate. This both makes it hard to distinguish between actually significant PRs and trivial ones and makes tests useful because no one reads the changes in them.

We should rethink which test files we're committing into source.

sampsyo, 2020-08-05:
Not a direct solution, but one way to make PRs a little easier to read is to filter out test files.
[image]

rachitnigam, 2020-09-10:
@sgpthomas let's try to fix this! I would start by removing verilog sources and inlined FuTIL for everything that is not an integration test.

Basically, compilation artifacts are only committed for small files and everything else is simulated.

rachitnigam, 2020-10-06:
Fixed by #228

closed 2020-10-06 by rachitnigam