calyxir/calyx #180: Broken timing for external memories

sgpthomas, 2020-08-04:
Turns out that for polybench benchmarks that write directly into externalized memories, we don't get timing reports for the parts of the circuits that write into these memories. This is bad because these parts of the circuit often don't meet timing and so our latency numbers are misleading.

This is what the Vivado warning looks like that tells us this is going to happen (this is the one in the methodology report, though there's a similar warning in the log as well):
```
TIMING-18#1 Warning
Missing input or output delay  
An input delay is missing on A_read_data[0] relative to clock(s) clk
```

We'll have to figure out how to give the externalized memory ports this input delay. I assume this is done in the `.xdc` file. I guess another option is to always write into a tmp variable before writing into a memory. This would slow us down though.

sampsyo, 2020-08-05:
Aha, that’s tricky... I don’t remember seeing anything in the .xdc for VHLS’s output, but it might be good to check if they’re doing something else for those memories? Happy to help dig into this.

sgpthomas, 2020-08-09:
Fixed in Dahlia by inserting registers in front of all reads and writes from/into memories

closed 2020-08-09 by sgpthomas