calyxir/calyx #183: Relay: Fix go/done signals and register write enables

sampsyo, 2020-08-10:
As mentioned in #182. A good first goal would be to make sure all the tests compile with the current FuTIL compiler; then we can add functional testing to make sure they produce the right outputs.

ViviYe, 2020-08-17:
I was trying to see if the generated futil files can compile 
```
   import "primitives/std.lib";
   component main() -> () {
     cells {
       x = prim std_reg(32);
       ret = prim std_reg(32);
       const0 = prim std_const(32, 10);
       add1 = prim std_add(32);
     }
     wires {
       group group2 {
         add1.left = x.out
         add1.right = const0.out
         ret.in = add1.out;
         ret.write_en = 1'd1
         group2[done] = ret.done;
       }
     }
 
     control {
       seq {
         group2
       }
     }
   }
```
somehow Futil compiler is yelling at me for
```
Error: FuTIL Parser:   --> 12:7
   |
12 |       add1.right = const0.out␊
   |       ^---
   |
   = expected guard_eq, guard_neq, guard_leq, guard_geq, guard_lt, guard_gt, guard_or, guard_and, or guard_not
```
and I couldn't figure out why : (

sampsyo, 2020-08-17:
My best guess is that it's the missing semicolon on the previous line?

rachitnigam, 2020-08-24:
@ViviYe is this working now? If so, can you close the issue? Otherwise, can you update us?

rachitnigam, 2020-10-06:
Trying to compile the programs under `frontend/relay-futil/tests/*.expect` using the FuTIL compiler. Ran into two errors:
```
⇒  fud e --from futil --to verilog frontends/relay-futil/tests/ifelse.expect
Stage 'futil' had a non-zero exit code.
--------------- stderr ---------------
Error:
30 |      res9.in = const7.out;
   |           ^^ [Papercut] Required signal not driven inside the group.
When driving the signal `res9.in' the signal `res9.write_en' must also be driven. The primitive type `std_reg' requires this invariant.
```

and 

```
⇒  fud e --from futil --to verilog frontends/relay-futil/tests/tensors_add.expect
Stage 'futil' had a non-zero exit code.
--------------- stderr ---------------
Error: FuTIL Parser:   --> 61:7
   |
61 |       while le0.out with cond0 {␊
   |       ^---
   |
   = expected stmt
```

sampsyo, 2020-10-07:
This might be a good place to leave a link to Jared's suggestion from Twitter:
https://twitter.com/roeschinc/status/1303840284304793602

He's got a proper Rust interface to TVM going. So once we figure out #199, maybe a direct translation from in-memory representation to in-memory representation would be possible, without going through the FuTIL text format.

closed 2020-10-12 by cgyurgyik