calyxir/calyx #198: Bitwidth Reduction

rachitnigam, 2020-08-24:
Several cells in the library have input and output bitwidths. A bitwidth reduction pass analyzes the uses of output ports and,
based on the use, changes the input bitwidths.

For example, we could have several 32-bit registers that feed a 32-bit adder which is directly connected to a 32-to-16 bit slice. This means that we could moves the slices to the input and turn the 32-bit adder into a 16-bit adder.

I don't have a specific idea for using such a pass and not sure how much effect it'll have in the end. However, we can keep thinking about it.

sampsyo, 2020-08-31:
While interesting, this pass has one tricky aspect to it: it would need to make assumptions about the individual components involved. For example, if a 32-bit adder is connected to a 16-bit slice, and we transform that to a 16-bit adder preceded by some slices to reduce the input bit-widths, that pass is assuming the two are semantically equivalent. That's true of adders but not guaranteed to be true of an arbitrary component. (As a contrived example, imagine a "parity + sign-extend" component, i.e., the output is all 0s iff the input has an even number of 0s.)

With that in mind, maybe it's worth considering whether this is best done in the frontend? I'd be interested to hear thoughts on what makes it less than entirely convenient to do right in Dahlia, for example.

rachitnigam, 2022-10-01:
It doesn't seem like we'll implement this anytime soon and not clear if its worth implementing. Closing!

closed 2022-10-01 by rachitnigam