calyxir/calyx #199: Redesigning Programmatic Interface

rachitnigam, 2020-08-24:
We've implemented a bunch of things and learned some things about the kind of things that the programmatic interface needs to be able to do.

Let's discuss our requirements and possible choices and implement things. There is a danger with discussions like these to spin out into abstract philosophizing without any action so let's put a **hard deadline of next Thursday (09/03)** so close this conversation and start implementing.

### Requirements

A programmatic interface has two requirements: analysis and manipulation. Here are some examples of things that each of these entail.

#### Analysis Tasks

In general, these seems like things that are best accomplished using a graph representation.

- Get all reads from a cell.
  - Get all reads from a port in a cell.
  - **NOTE**: Reads from ports in a guard are not currently considered as reads.
- Get all writes to a cell/port.
- Get all reads/writes in a group.

#### Manipulation Tasks

These seem like these would be best accomplished using some form of "structural" representation of the AST: Something like a list of assignments and cells.

- Create a new guarded assignment statement in a group.
  - In the current implementation, when we create a assignment it fells like we're manipulating a graph. It seems that creation of and manipulation should not be tied to the graph representation.
- Change a guard/destination port/source port in an assignment.

#### Bonus: Sharing Information

We've not yet built passes that share information between each other. Perhaps this is best accomplished as separate data structures but those would like either like graphs or hash maps or something.

#### Together, at last

The hard part of coming with a representation is that most passes perform both analysis and manipulation.
For example, the inlining pass needs to collect all reads to a hole (analysis), create a new expression, and inline the expression into all the uses (manipulation).

### Possible designs

1. [LNAST](https://woset-workshop.github.io/PDFs/2019/a3.pdf): It uses two representations at the same time. One that is a graph and another that is structural.
2. LLVM: I don't have the best idea about this. Maybe @sampsyo can add some thoughts?

sampsyo, 2020-08-24:
Sounds great! The programmatic interface to the IR is definitely the part of FuTIL I know the least about. It would be great to talk about it some more.

The only commentary I have about "the LLVM way" is this: in LLVM, the instruction-oriented and graph-oriented representations are one and the same, which is a pretty nifty thing enabled by SSA. That is, `Instruction` objects (and `Value` objects more generally) point directly to each other to form a DFG (and other graphs more generally, including a CFG) in addition to existing in a sequential order in a containment hierarchy (modules contain functions contain BBs contain instructions). So it is very convenient to traverse the same data structure in two different ways: linearly through time, and graphically through dependence edges.

A consequence is that the translation between the in-memory data structure and the on-disk IR is somewhat fancy.

rachitnigam, 2020-10-05:
## Redesigning the frontend data structures

Observations and recommended changes for the frontend data structures. 
There are a few problems with the current data structures
- **Mixing AST and data structures**: There are certain AST nodes used as data structures. This is bad idea---ASTs are supposed to capture minute differences between all AST nodes; a program representation should homogenize irrelevant details.
- **No separation of program context**: There is a lot of information, such as signatures for nodes and components, that is attached to the data structures. Instead of doing this, we should separate out the program context information such as signatures and make the internal data structures leaner.
- **Constants**: Constants are magical and therefore don't have a standard representation. Sometimes, they are "just" nodes and sometimes they are magically memoized.
- **Guards are ASTs**: This is bad idea because the ports in the guards are not regarded as "reads" by the graph representation.

I have lower level comments. *My comments are italicized*.

### Main Data Structures

- **Context**: The top-level data structure that (seems to) contain everything.
  - **Library Context**: Information about the libraries. AFAICT, this is only used during code generation.
  - **Definitions**: Map from names to components. The names are definition names and *not* instance names. This contains information both for code generation and program representation. 
    - **Component**

*Why is it necessary to store the components in the context The context seems to serve as both the "top-level" data structure and the source of context information. It doesn't have to do this.*
      
- **Component**:  A FuTIL component. Contains the name, signature, control structures, connection graph and signature for resolved compoents.
  - **Resolved Signatures**: Seemingly contains the signature resolution for all the used components. *I would have expected this to be a part of library context*.
  - **Signature**: The signature of this component. *I would again expect this to be in the library context*.
  - **Control**: The control program associated with this component.
  - **StructureGraph**

- **StructureGraph**: A graph-based representation of the program. Contains far too much information.
  - **Constants**: Memoized nodes for each constant. *This memoization/canonicalization should be done before the structure graph is generated*.
  - **I/O**: A fake node to represent the current component.
  - **Groups**: Map from name of a group to references to the edges contained in it.
  - **Graph**: The program graph. Contains **EdgeData** and **Node**.

- **EdgeData**: The data associated with each "edge". An edge is created for assignments (`c.in = foo & bar ? d.out`).
  - **Guard**: An *AST* that represents the guard for this assignment. *Using an AST for the guard is bad idea for two reasons: (1) constructing it requires annoying AST construction, and (2) the graph does not capture use of ports as "reads" from a node. The latter is a really big problem.*.
  - **Width**: Bitwidth for the edge. *This information should be retrievable from the program context and not be attached to an edge. For example, if I know that one side of the edge is `r.in`, I should be able to query `r` and get the width of `in`.*
  - **Group**: Group with which this edge is associated. *This information is duplicated. The `groups` part of the StructureGraph.*

- **Node**: Confusingly named data structure carrying node information. *A better name would be `NodeData` which is already taken by a different thing.*
  - **Signature**: The input and output ports of this node. *This information should be in the program context.*
  - **NodeData**: "Tag" information about this node. Used to distinguish it as a compilation hole, **Cell**, constant, or port on this component

- **Cell**: Tag for most Nodes. Represents a declared node or a primitive.
  - **Prim**: A primitive node.
    - **Compinst**: The information required to initialize this primitive.
  - **Decl**: A FuTIL component
*This hierarchy should be collapsed after parsing. There is no reason `Decl` can't have an empty Compinst*.

sgpthomas, 2020-10-06:
Several clarifications:
 - The library context is also used to instantiate primitive components during a pass. If you want to add a new component from a primitive library, then you need to specialize the signature using parameters. The library context lets you do this. Refer [here](https://github.com/cucapra/futil/blob/e1eb074d0d2aa5f22b02ee5444323949e95dbda8/calyx/src/passes/remove_external_memories.rs#L60) for an example.
 - Resolved signatures are in the component and not in the library context because they contain specialized signatures for this component definition. The library context has parameterized signatures and component has relevant specialized signatures.

I realized that the main reason for having the context own the component definitions is because it was the easiest way I could find to give passes mutable access to the signatures to the visitor as well as immutable access to the context.
In it's current form, the component needs mutable access to the signature so that passes can add / remove ports for the signature.

The context needs immutable access to the signature of all components so that passes can look up the signatures of other components.

However, the visitor already owns the mutable access to the component, and so the context can't take out immutable references to the signature. 

We can get around this problem by having the context own the components and then use `RefCells` to illegally pass out mutable access to the visitor.

However, I think the problem goes away if we separate the context and the component ownership and have the context store clones of the signatures, rather than a reference to them. The visitor would have to keep track of changes to the signature and then update the context with the new signature after the visitor is done.

rachitnigam, 2020-10-06:
> The library context is also used to instantiate primitive components during a pass. If you want to add a new component from a primitive library, then you need to specialize the signature using parameters. The library context lets you do this. Refer here for an example.

Can we be clearer with the cell vs. component distinction? Are you saying that a pass could instantiate a new cell and that requires a context. If so, I agree. However, this doesn't mean that the context needs to own the components.

> Resolved signatures are in the component and not in the library context because they contain specialized signatures for this component definition. The library context has parameterized signatures and component has relevant specialized signatures

Sure. I was pointing out that even within a Component, this information doesn't need to be embedded within nodes. It seems like a Component might want a local Context that remembers the "type signature" (all the parameters needed for specialization) for each cell.

> In it's current form, the component needs mutable access to the signature so that passes can add / remove ports for the signature.

I want to thing really hard about this decision. Mutating a component's signature is a global and invasive change. I think the current way of doing this is quite baroque: when the compilation process start, each component magically gets a `go` and `done` signal added to its interface. Eventually, the inlining pass refies these signals. This seems like the wrong approach. I'm not convinced that changing signatures is such a good idea.

> However, I think the problem goes away if we separate the context and the component ownership and have the context store clones of the signatures, rather than a reference to them. The visitor would [...]

This problem goes away if signatures are immutable.

PR #246 (2020-10-23): Frontend Redesign
Redesigns the FuTIL compiler's internal representation (IR). Because this representation is pervasively used, the PR touches basically everything. The high-level changes are:

- A new IR under `calyx::ir`.
  - `calyx::traversal` exposes a visitor-based infrastructure to traverse control programs.
  - `calyx::analysis` implements a graph-based analysis that calculates the reads-from and writes-to sets.
- Consolidation of frontend code.
  - Most of the AST defined in `lang::ast` is now under `frontend::ast`. The PR simplified some of the AST nodes.

## TODO
- [ ] ~Port `merge-a [... cut]
closed 2020-10-23 by rachitnigam