calyxir/calyx #200: Control operator for Pipelining

rachitnigam, 2020-09-02:
Starting a tracker for implementing pipelining. We should start with simple "feed-forward" pipelines which don't the ability to apply an back-pressure (too restrictive @sampsyo?). 

We can start with the following observation: Pipelines can be currently represented in FuTIL. It is just a `while` with nested `par` and some extra logic to decide when the pipeline is done. This means that we could think about pipelining as a "level 2" operator that is compiled away to primitive control operators.

### Representing a pipeline

The execution of a pipeline proceeds in three phases:
1. Filling up the pipeline: Stages towards the end of pipeline don't have values yet.
2. Steady-state execution: Every stage has a value.
3. Flushing: New inputs have stopped appearing and the stages in the front are not processing values.

A simple way to represent this is the following control:
```
seq {
  // phase 1
  par { stage_1 }
  par { stage_1; stage_2 }
  ...
  // phase 2
  while steady.out with steady_check {
    par { stage_1; stage_2; ... stage_n }
  }
  // phase 3
  par { stage_2; stage_3; ... stage_n }
  par { stage_3; ... stage_n }
  ...
}
```

### Discussion questions

To start off the discussion, we have to answer some questions about the source level representation of pipelines.

1. How are buffers between pipeline stages represented? They can either be compilation holes that are reified during compilation or they could be explicit like in the case of the systolic array.
2. (Related to 1) What should be the interface for groups in a pipeline? Do we need to add new holes to make them have stronger interfaces?
3. What does the `go`-`done`/communication interface look like? Pipelines can take inputs every cycle so how should the `go`-`done` interface work. We can either try to overload the meaning of go and done or try to come up with new signals that signal to the context that the pipeline is ready to execute.

sampsyo, 2020-09-02:
Awesome. I think starting with simple feed-forward pipelines is the right thing to do—it presents plenty of intellectual challenges by itself, and it suffices to represent plenty of actually-useful architectures. And the "level 2" control operator that can be lowered into further control statements seems like a good idea to me too. It's slightly off topic, but the same approach might work for unrolling-like duplication for parallelism.

I think a good way to go would be to sketch out the "hole-based" interface between pipeline stages. The thing we need, IMO, is a way to represents the "inputs" and "outputs" of a stage, which we don't really have for other groups. But it might come in handy for other stuff too.

sgpthomas, 2020-10-13:
https://people.csail.mit.edu/rinard/paper/isss01.pdf This is a paper I found that describes an optimization to take a computation that is crudely pipelined and automatically extract more stages to more efficiently pipeline the computation. This optimization may be enabled by having an explicit pipelining control operator.

sampsyo, 2020-10-14:
Huh, super interesting—good find, @sgpthomas. Also interesting that it does forwarding & speculation as well, on top of basic pipelining. Seems like @dz333 might be interested as well if he hasn't already seen it?

dz333, 2020-10-14:
Oh interesting - that paper is based off of a proto-bluespec like language
(i.e. it has a set of rules that all try to 'atomically' fire every cycle if their conditions are met - which can be translated to RTL through term-rewriting).

Their algorithm (ignoring the part where they insert a bunch of forwarding/speculation) doesn't actually assume anything about the existing circuit - they just enforce that the newly generated circuit has a specific interface between modules (stages). So I don't see why you would really need a dedicated 'pipeline control' operator. Instead you would need to determine what are the boundaries that you can't pipeline across in your control language. 

E.g. in their language "pipelining" serves to take 1 rule and split it into multiple. I'm not 100% sure what that would be in FuTIL but probably something along the lines of taking a `par` block and turning it into a `seq { par , par }`

rachitnigam, 2021-03-04:
Playing around with some syntax, the pipeline operator could look like this:

```
pipeline (left, right, acc)(out = a.out) {
  stage m(out) {
    mult.go = 1;
    mult.left = left;
    mult.right = right;
    m.out = mult.out;
    m[done] = mult.done;
  }
  stage a(out) {
    add.go = 1;
    add.left = acc;
    add.right = m.out;
    a.out = add.out;
    a[done] = add.done;
  }
}
```

The stage wrappers look *almost* similar to groups but they take in extra inputs. One way to unify groups and stages is allowing groups to take in inputs, in which case stages just become groups. A different approach can be replacing stages with real components which will also allow stages to use arbitrary control.

In order to compile this description, we'll have two options:
1. Latency-sensitive compilation: If all the stages are timed, we should be able to generate independent FSMs for each stage and connect them using appropriately sized buffers.
2. Latency-insensitive compilation: If the stages are untimed, the compiler needs to generate ready-valid signals between each stage. The ready signal for a stage is its done signal while the valid signal is the `go` signal. We need to be careful about this interpretation of the signals because done signals are only held up for one cycle. This might be different from how a traditional ready-valid interface works.

dz333, 2021-03-05:
Hmmmm, I'm not entirely sure what the value of adding these to Calyx is (as opposed to just having groups). What does adding input parameters to groups really buy you (genuinely asking since I don't 100% understand them)?

In response to your latency-insensitive point:

You probably can't just leave the `done` signal (a.k.a. the `ready` signal) high for one cycle unless you can be sure that there is _always_ valid data. Normally `ready` is high until the cycle when `ready & valid` is high, at which point the data is transferred and then `ready` may go down or not on the next cycle (depending on the pipeline implementation).

rachitnigam, 2022-02-11:
We've shown how to implement a pipeline in Calyx with the PR above and folks in the CIRCT group have been using it to generate it in Calyx. I'm going to close this issue now and revisit it in the future when we want to optimize Calyx pipelines!

closed 2022-02-11 by rachitnigam