calyxir/calyx #235: A lone if doesn't compile correctly

sgpthomas, 2020-10-05:
```C
import "primitives/std.lib";

component main() -> () {
  cells {
    mem = prim std_mem_d1(32, 1, 1);
    lt = prim std_lt(32);
  }

  wires {
    group cond {
      lt.left = 32'd5;
      lt.right = 32'd9;
      cond[done] = 1'b1;
    }

    group true {
      mem.addr0 = 1'b0;
      mem.write_data = 32'b1;
      mem.write_en = 1'b1;
      true[done] = x.done;
    }

    group false {
      mem.addr0 = 1'b0;
      mem.write_data = 32'b0;
      mem.write_en = 1'b1;
      false[done] = y.done;
    }
  }

  control {
    if lt.out with cond {
      true;
    } else {
      false;
    }
  }
}
```

This program causes a clk loop:
```
%Warning-UNOPTFLAT: /tmp/tmpoqb_96re:61:9: Signal unoptimizable: Feedback to clock or circular logic: 'main.lt_out'
   61 |   logic lt_out;
      |         ^~~~~~
                    ... Use "/* verilator lint_off UNOPTFLAT */" and lint_on around source to disable this message.
                    /tmp/tmpoqb_96re:61:9:      Example path: main.lt_out
                    /tmp/tmpoqb_96re:138:3:      Example path: ALWAYS
                    /tmp/tmpoqb_96re:59:16:      Example path: main.lt_left
                    /tmp/tmpoqb_96re:25:14:      Example path: ASSIGNW
                    /tmp/tmpoqb_96re:61:9:      Example path: main.lt_out
%Error: Exiting due to 1 warning(s)
```

sgpthomas, 2020-10-05:
```verilog
/* verilator lint_off WIDTH */
module std_reg
  #(parameter width = 32)
   (input wire [width-1:0] in,
    input wire write_en,
    input wire clk,
    // output
    output logic [width - 1:0] out,
    output logic done);

  always_ff @(posedge clk) begin
    if (write_en) begin
      out <= in;
      done <= 1'd1;
    end else
      done <= 1'd0;
  end
endmodule

module std_lt
  #(parameter width = 32)
  (input logic [width-1:0] left,
    input logic [width-1:0] right,
    output logic            out);
  assign out = left < right;
endmodule

module std_mem_d1
  #(parameter width = 32,
    parameter size = 16,
    parameter idx_size = 4)
   (input logic [idx_size-1:0] addr0,
    input logic [width-1:0]   write_data,
    input logic               write_en,
    input logic               clk,
    output logic [width-1:0]  read_data,
    output logic done);

  logic [width-1:0]  mem[size-1:0];

  assign read_data = mem[addr0];
  always_ff @(posedge clk) begin
    if (write_en) begin
      mem[addr0] <= write_data;
      done <= 1'd1;
    end else
      done <= 1'd0;
  end
endmodule

// Component Signature
module main (
      input logic go,
      input logic clk,
      output logic done
  );
  
  // Structure wire declarations
  logic [31:0] lt_left;
  logic [31:0] lt_right;
  logic lt_out;
  logic mem_addr0;
  logic [31:0] mem_write_data;
  logic mem_write_en;
  logic mem_clk;
  logic [31:0] mem_read_data;
  logic mem_done;
  logic cond_computed0_in;
  logic cond_computed0_write_en;
  logic cond_computed0_clk;
  logic cond_computed0_out;
  logic cond_computed0_done;
  logic cond_stored0_in;
  logic cond_stored0_write_en;
  logic cond_stored0_clk;
  logic cond_stored0_out;
  logic cond_stored0_done;
  logic done_reg0_in;
  logic done_reg0_write_en;
  logic done_reg0_clk;
  logic done_reg0_out;
  logic done_reg0_done;
  
  // Subcomponent Instances
  std_lt #(32) lt (
      .left(lt_left),
      .right(lt_right),
      .out(lt_out)
  );
  
  std_mem_d1 #(32, 1, 1) mem (
      .addr0(mem_addr0),
      .write_data(mem_write_data),
      .write_en(mem_write_en),
      .clk(clk),
      .read_data(mem_read_data),
      .done(mem_done)
  );
  
  std_reg #(1) cond_computed0 (
      .in(cond_computed0_in),
      .write_en(cond_computed0_write_en),
      .clk(clk),
      .out(cond_computed0_out),
      .done(cond_computed0_done)
  );
  
  std_reg #(1) cond_stored0 (
      .in(cond_stored0_in),
      .write_en(cond_stored0_write_en),
      .clk(clk),
      .out(cond_stored0_out),
      .done(cond_stored0_done)
  );
  
  std_reg #(1) done_reg0 (
      .in(done_reg0_in),
      .write_en(done_reg0_write_en),
      .clk(clk),
      .out(done_reg0_out),
      .done(done_reg0_done)
  );
  
  // Memory initialization / finalization 
  import "DPI-C" function string futil_getenv (input string env_var);
  string DATA;
   initial begin
  DATA = futil_getenv("DATA");
      $fdisplay(2, "DATA (path to meminit files): %s", DATA);
      $readmemh({ DATA, "/mem.dat" }, mem.mem);
  end
  
   final begin
      $writememh({ DATA, "/mem.out" }, mem.mem);
  end
  
  // Input / output connections
  always_comb begin
    if (done_reg0_out)
      done = 1'd1;
    else 
      done = '0;
    
    if ((!cond_computed0_out & go))
      lt_left = 32'd5;
    else 
      lt_left = '0;
    
    if ((!cond_computed0_out & go))
      lt_right = 32'd9;
    else 
      lt_right = '0;
    
    if ((!mem_done & cond_computed0_out & cond_stored0_out & go | !mem_done & cond_computed0_out & !cond_stored0_out & go))
      mem_addr0 = 1'd0;
    else 
      mem_addr0 = '0;
    
    if ((!mem_done & cond_computed0_out & !cond_stored0_out & go))
      mem_write_data = 32'd0;
    else if ((!mem_done & cond_computed0_out & cond_stored0_out & go))
      mem_write_data = 32'd32;
    else 
      mem_write_data = '0;
    
    if ((!mem_done & cond_computed0_out & cond_stored0_out & go | !mem_done & cond_computed0_out & !cond_stored0_out & go))
      mem_write_en = 1'd1;
    else 
      mem_write_en = '0;
    
    if (done_reg0_out)
      cond_computed0_in = 1'd0;
    else if ((!cond_computed0_out & go))
      cond_computed0_in = 1'd1;
    else 
      cond_computed0_in = '0;
    
    if ((!cond_computed0_out & go | done_reg0_out))
      cond_computed0_write_en = 1'd1;
    else 
      cond_computed0_write_en = '0;
    
    if ((!cond_computed0_out & go))
      cond_stored0_in = lt_out;
    else if (done_reg0_out)
      cond_stored0_in = 1'd0;
    else 
      cond_stored0_in = '0;
    
    if ((!cond_computed0_out & go))
      cond_stored0_write_en = lt_out;
    else if (done_reg0_out)
      cond_stored0_write_en = 1'd1;
    else 
      cond_stored0_write_en = '0;
    
    if (done_reg0_out)
      done_reg0_in = 1'd0;
    else if (((cond_computed0_out & cond_stored0_out & mem_done | cond_computed0_out & !cond_stored0_out & mem_done) & go))
      done_reg0_in = 1'd1;
    else 
      done_reg0_in = '0;
    
    if (((cond_computed0_out & cond_stored0_out & mem_done | cond_computed0_out & !cond_stored0_out & mem_done) & go | done_reg0_out))
      done_reg0_write_en = 1'd1;
    else 
      done_reg0_write_en = '0;
    
  end
endmodule // end main
```

sgpthomas, 2020-10-05:
strangely if you change the control to
```
seq {
  if lt.out with cond {
      true;
    } else {
      false;
    }
}
```
it works fine. There doesn't seem to be anything different about how `lt` is wired so I don't know whats up here.

rachitnigam, 2020-10-05:
I had a guess: The generate if-else merge broke something.

I did a `git bisect` and found the offending commit (no surprises):
669f55acd8cf705b3cdf607d79cef3920ba16027 is the first bad commit

commit 32cb7746 (2020-12-09): if compilation magically started working again. Fixes #235
PR #287 (2020-12-22): Use assign instead of always_comb
Fixes #235. Fixes #285.
closed 2020-12-09 by rachitnigam