calyxir/calyx #240: [papercut] using an component in control has a bad error message

sgpthomas, 2020-10-07:
Using a component defined in `cells` instead of a group in control gives a bad error message. Here's an example program:
```
component main() -> () {
  cells {
    save = prim std_reg(32);
  }
  wires {}
  control {
    save;
  }
}
```
and the error message:
```
thread 'main' panicked at 'port!: Use of undefined input/output port: go', calyx/src/passes/component_interface.rs:34:13
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

rachitnigam, 2020-10-07:
This is the error message generated on master:
```
Error:
9 |    save;
   |    ^^^^ Use of undefined group: save
```

Not sure where you're getting the runtime error from.

sgpthomas, 2020-10-07:
ah, it just happens with `-d no-inline`. The alias probably doesn't run the relevant pass.

closed 2020-10-07 by rachitnigam