calyxir/calyx #264: Invoking multi-cycle components broken

rachitnigam, 2020-10-27:
The following program returns the wrong results:
```
import "primitives/std.lib";
component main() -> () {
  cells {
    x = prim std_reg(32);
    y = prim std_reg(32);
    mult_pipe0 = prim std_mult_pipe(32);
  }
  wires {
    group let3<"static"=4> {
      mult_pipe0.left = 32'd10;
      mult_pipe0.right = 32'd20;
      mult_pipe0.go = !mult_pipe0.done ? 1'd1;

      x.in = mult_pipe0.out;
      x.write_en = mult_pipe0.done;

      let3[done] = x.done;
    }
    group let4<"static"=4> {
      mult_pipe0.left = 32'd30;
      mult_pipe0.right = 32'd40;
      mult_pipe0.go = !mult_pipe0.done ? 1'd1;

      y.in = mult_pipe0.out;
      y.write_en = mult_pipe0.done;

      let4[done] = y.done;
    }
  }
  control {
    seq {
      let3;
      let4;
    }
  }
}
```

The offending lines are the guards to the `go` signals of the `mult_pipe0`
units. Since the guards say:
```
mult_pipe0.go = !mult_pipe0.done ? 1'd1;
```
The `go` signal is set to one cycle after the `done` is raised.

Normally, this is not a problem because the group usually finishes executing.
However, in this case, we assign the value in the multiplier to a
register which causes the group to take one extra cycle and re-enables the
multiplier.
Next, when the `let4` group executes, the multiplier's `go` signal has already
been high for one cycle.
This means that the `done` signal for the multiplier is prematurely raised
in `let4` causing the incorrect value to be written to `y`.

We didn't catch this issue because the Dahlia frontend never shares
multipliers.
It was uncovered by the resource sharing pass.

In general, this is going to be a problem for groups that enable multi-cycle
components (even FuTIL) ones since the group usually doesn't make sure that
the component only runs once.

One possible solution is changing the frontends to generate additional
registers to make sure that the `go` signal is only raised once in the
execution of the group but that feels somewhat unclean.

rachitnigam, 2020-10-28:
RFC from @sampsyo @sgpthomas

sampsyo, 2020-10-28:
Just wanted to let you know that I have seen this ping & added it to my queue but I am light-years behind during this crazy week and it will take me a bit to get to having useful thoughts! I will probably need to think more to understand what this means:

> However, in this case, we assign the value in the multiplier to a
> register which causes the group to take one extra cycle and re-enables the
> multiplier.
> Next, when the `let4` group executes, the [...]

I don't yet understand why this is the case but this is due to my deadline-addled brain not being able to focus on things this week.

rachitnigam, 2020-10-29:
Sounds good. Here is one more attempt to explain what's going on. In both `let3` and `let4`, we do the following:

- cycle 0: read some values and set `mult_pipe0 = 1` (`!mult_pipe0.done == 0`)
- cycle 0..3: Get the value from `mult_pipe0` and set `mult_pipe0.go = 0` (`!mult_pipe0.done == 1`).
- cycle 3..4: Save the value from `mult_pipe0` into a register.
                   : (*Undesired behavior*) Set `mult_pipe0.go = 1` (`!mult_pipe0.done == 0`) by mistake!!

Since `let3` and `let4` run in sequence, the mistaken `mult_pipe0.go = 1` in `let3` causes `let4` to see the wrong values.

sampsyo, 2020-10-29:
Ah, I finally understand! Thank you for this helpful breakdown! The insight I was missing is that the offending assignment is firing because our “protocol” for how to “invoke” components fails to account for time spent *after* the invoked component completes—because its done signal only stays high for a moment, the group observes it becoming low again, and then says, “aha, it must be time to re-invoke this component!!”

One thing I would like to ponder is whether this problem also exists in code generated from lowering `seq`... that is, if we had code like this:

```
seq {
  seq { mult ; save_in_x };
  seq { mult ; save_in_y }
}
```

(assuming maybe-obvious definitions of the groups involved), and we compiled the inner `seq`s, we would generate code that has this issue, right?

rachitnigam, 2020-12-26:
This will be fixed #305.

closed 2020-12-26 by rachitnigam

rachitnigam, 2021-03-23:
> This means that the done signal for the multiplier is prematurely raised in let4 causing the incorrect value to be written to y.
This diagnosis was wrong—the done signal is clobbered when `go` is 0 for one or more cycles. The actual problem was that the static timing pass was generating the wrong FSM transition for the first state.