calyxir/calyx #267: Relay2Futil implementation discussion

cgyurgyik, 2020-10-28:
## Problem
I found myself focusing on minute aspects when implementing Relay2Futil, such as trying to get 
```
v0.0.4
fn (%a: int32) {
  %a
}
``` 
to work coherently with other functions. I then realized that maybe this wasn't a good use of time, and decided to take a step back. Let's look at the Relay representation of the test program [MLP Net](https://tvm.apache.org/docs/api/python/relay/testing.html#tvm.relay.testing.mlp.get_net):
```
fn (%data: Tensor[(1, 1, 28, 28), float32], %fc1_weight: Tensor[(128, 784), float32], %fc1_bias: Tensor[(128), float32], %fc2_weight: Tensor[(64, 128), float32], %fc2_bias: Tensor[(64), float32], %fc3_weight: Tensor[(10, 64), float32], %fc3_bias: Tensor[(10), float32]) -> Tensor[(1, 10), float32] {
  let %x: Tensor[(1, 784), float32] = nn.batch_flatten(%data) /* ty=Tensor[(1, 784), float32] */;
  let %x1: Tensor[(1, 128), float32] = nn.dense(%x, %fc1_weight, units=128) /* ty=Tensor[(1, 128), float32] */;
  let %x2: Tensor[(1, 128), float32] = nn.bias_add(%x1, %fc1_bias, axis=-1) /* ty=Tensor[(1, 128), float32] */;
  let %x3: Tensor[(1, 128), float32] = nn.relu(%x2) /* ty=Tensor[(1, 128), float32] */;
  let %x4: Tensor[(1, 64), float32] = nn.dense(%x3, %fc2_weight, units=64) /* ty=Tensor[(1, 64), float32] */;
  let %x5: Tensor[(1, 64), float32] = nn.bias_add(%x4, %fc2_bias, axis=-1) /* ty=Tensor[(1, 64), float32] */;
  let %x6: Tensor[(1, 64), float32] = nn.relu(%x5) /* ty=Tensor[(1, 64), float32] */;
  let %x7: Tensor[(1, 10), float32] = nn.dense(%x6, %fc3_weight, units=10) /* ty=Tensor[(1, 10), float32] */;
  let %x8: Tensor[(1, 10), float32] = nn.bias_add(%x7, %fc3_bias, axis=-1) /* ty=Tensor[(1, 10), float32] */;
  let %x9: Tensor[(1, 10), float32] = nn.softmax(%x8) /* ty=Tensor[(1, 10), float32] */;
  %x9
}
```

Here, we're simply passing in the intermediary result of function `i` into function `i+1`. So, I decided to instead focus on this pattern.

## Design
1. I've decided to use `std_memory_d*` to represent everything, to include scalars. This allows me to use runt to quickly test the expected outcome of a program. It also will make the transition to tensors of dimension 1 or greater a bit smoother as I've become more familiar with passing memory "by reference." One thing Rachit mentioned in our 1:1 is looking into the simplification of this.

2. The proposed outcome is the following: if a user wants to use a new function `foo`, he or she will need to implement the body of said function, and from there add the name to `visit_call`, and ideally, it should work. However, this isn't totally clear yet. For example, I need to put more thought into how to approach N-dimensional tensors, where N > 0. For example, if I want to do an `add` on two vectors, should I simply call `add` N times in `main`, or should I pass in `begin, end` constants to represent the first and last index? I'm leaning toward the latter, but we'll see.

3. The code needs clean up! Given this has been mostly experimental with many design iterations on my end, expect funky code.

## Example
Let's take the following nested function example:
```
v0.0.4
fn (%a: int32, %b: int32) {
  let %c = multiply(%a, %b);
  let %d = add(%c, %a);
  %d
}
```

I've currently implemented 0-dimensional tensor binary operations for the supported primitive operations. This means that in ```visit_call(self, call)```, I can simply return `build_tensor_0D_binary_op(call, args, op)` (Assuming the operands are legal and `op` a supported operation). We will represent the Relay function `fn` as a component, from which our binary operations will be called. `%c, %d` are two intermediary results that will also be saved in a 1x1 `std_memory_d1`. Each sub-function will be called, and then a closing group will save the return value in `main`, which will run the Relay function.

```
import "primitives/std.lib";

component mult (a_out: 32, b_out: 32, in_done: 1) -> (in_write_data: 32, in_write_en: 1, in_addr0: 1) {
  cells {
    mult = prim std_mult(32);
    c0 = prim std_const(1, 0);
  }
  wires {
    group process_mult {
      in_addr0 = c0.out;
      mult.left = a_out;
      mult.right = b_out;
      in_write_en = 1'd1;
      in_write_data = mult.out;
      process_mult[done] = in_done ? 1'd1;
    }
  }
  control {
    seq {
      process_mult;
    }
  }
}
component add (c_out: 32, a_out: 32, in_done: 1) -> (in_write_data: 32, in_write_en: 1, in_addr0: 1) {
  cells {
    add = prim std_add(32);
    c0 = prim std_const(1, 0);
  }
  wires {
    group process_add {
      in_addr0 = c0.out;
      add.left = c_out;
      add.right = a_out;
      in_write_en = 1'd1;
      in_write_data = add.out;
      process_add[done] = in_done ? 1'd1;
    }
  }
  control {
    seq {
      process_add;
    }
  }
}
component function0 (a_out: 32, b_out: 32, in_done: 1) -> (in_write_data: 32, in_write_en: 1, in_addr0: 1) {
  cells {
    add_fn = add;
    mult_fn = mult;
    c = prim std_mem_d1(32, 1, 1);
    d = prim std_mem_d1(32, 1, 1);
    c0 = prim std_const(1, 0);
  }
  wires {
    group run_mult_fn {
      mult_fn.a_out = a_out;
      mult_fn.b_out = b_out;
      c.write_data = mult_fn.in_write_data;
      c.write_en = mult_fn.in_write_en;
      c.addr0 = mult_fn.in_addr0;
      mult_fn.in_done = c.done;
      mult_fn.go = 1'd1;
      run_mult_fn[done] = mult_fn.done ? 1'd1;
    }
    group run_add_fn {
      add_fn.c_out = c.read_data;
      add_fn.a_out = a_out;
      d.write_data = add_fn.in_write_data;
      d.write_en = add_fn.in_write_en;
      d.addr0 = add_fn.in_addr0;
      add_fn.in_done = d.done;
      add_fn.go = 1'd1;
      run_add_fn[done] = add_fn.done ? 1'd1;
    }
    group save_return_value {
      d.addr0 = c0.out;
      in_addr0 = c0.out;
      in_write_en = 1'd1;
      in_write_data = d.read_data;
      save_return_value[done] = in_done ? 1'd1;
    }
  }
  control {
    seq {
      run_mult_fn;
      run_add_fn;
      save_return_value;
    }
  }
}

component main () -> () {
  cells {
    fn0 = function0;
    c0 = prim std_const(1, 0);
    main_ret = prim std_mem_d1(32, 1, 1);
    a = prim std_mem_d1(32, 1, 1);
    b = prim std_mem_d1(32, 1, 1);
  }
  wires {
    group run_fn0 {
      fn0.a_out = a.read_data;
      a.addr0 = fn0.in_addr0;
      fn0.b_out = b.read_data;
      b.addr0 = fn0.in_addr0;
      main_ret.addr0 = fn0.in_addr0;
      main_ret.write_data = fn0.in_write_data;
      main_ret.write_en = fn0.in_write_en;
      fn0.in_done = main_ret.done;
      fn0.go = 1'd1;
      run_fn0[done] = fn0.done ? 1'd1;
    }
  }
  control {
    seq {
      run_fn0;
    }
  }
}
```

## Closing Thoughts
As mentioned above, I still need to think about vectors, matrices, and so on, so this may throw a few wrenches in the simplicity of this plan. This also does not mean that other aspects of the Relay IR (such as if statements) are off limits -- I just haven't gotten to them. With that said, I haven't touched passing in constants yet as function arguments, so I'll need to work on this as well. I'm happy to hear any criticisms, thoughts, or questions. This was mostly a byproduct of frustratingly minimal progress being made, so maybe there's better approaches you can think of, which I'd be glad to hear of.

cgyurgyik, 2020-10-29:
Discussed today: Bring Dahlia into the picture. Instead of writing our functions such as `add`, `relu`, etc. manually using FuTIL, let's have Dahlia conduct the hard labor, and then we can just link these together using the Relay2FuTIL compiler.

More specifically, I initially intended for, say, `nn.relu` to be written using FuTIL. However, we should write this using the higher level language Dahlia, lower it to FuTIL, and then drop it in as a component named `relu` for use where appropriate. This should of course apply to all functions. 

@rachitnigam Can you check my understanding?

sampsyo, 2020-10-29:
The relevant issue from the past is #186, FWIW.

rachitnigam, 2020-10-29:
The understanding is correct!

cgyurgyik, 2020-10-29:
> The relevant issue from the past is #186, FWIW.

> The understanding is correct!

Thanks and thanks!

closed 2020-11-11 by cgyurgyik