calyxir/calyx #274: [fud] Add named components for Dahlia lowering to fud, FuTIL lowering.

cgyurgyik, 2020-11-12:
Currently, with the Dahlia binary, we can do the following: 
```
./fuse foo.fuse --lower -b=futil -n=<component_name>
```
We'd like to be able to do the same in fud, so that we can use fud directly in the Relay to FuTIL lowering.

On a similar note, currently trying to run a pass on FuTIL code in fud gives an error. For example:
```
fud e --from futil program.futil --to futil -s futil.flags '-p externalize'
```
will lead to an error along the lines of: 
```ends at the same stage `futil'. This is likely an error```

This should be a simple fix as well, though maybe there's a reason this is enforced.

rachitnigam, 2020-11-25:
1. Fixed in 5a9513dd25cbc6634641547775f0ea35fe404902
2. Fud operates on a DAG of named stages. If we allowed transformations from a step to itself, it's no longer a DAG. See main.py to see how stages like `futil-lowered` are registered. You can create a new registry with those flags.

rachitnigam, 2020-12-15:
@cgyurgyik can you respond to this?

cgyurgyik, 2020-12-15:
I've registered a new stage in #289 (linked) for FuTIL lowering using `-p externalize`. I didn't touch the named components in Dahlia. Right now, I just use the `fuse` binary.

PR #289 (2020-12-20): [Relay] softmax, max_pool2d, conv2d, fix 4D memory connections (WIP)
### Overview
- `nn.max_pool2d` implementation with test.
- `nn.conv2d` implementation with test.
- Fixes to 4D to actually hook in all the appropriate addresses. This includes a `runt` test, since I missed this bug in an earlier PR.
- Fixes #274 .
- Improves modularity. Everything that deals with `Relay function call` -> `Dahlia function` -> `FuTIL lowering` is done in a single file `dahlia_lowering.py` now.
- Allows for lowering of `VGG Net` from Relay IR to FuTIL, i.e. `python3 example.py vgg_net`. This is by no means fast; [... cut]
closed 2020-12-20 by cgyurgyik