calyxir/calyx #276: [Relay] Steps left for functioning VGG Net lowering.

cgyurgyik, 2020-11-12:
I spent some time looking over the [VGG](https://github.com/apache/incubator-tvm/blob/main/python/tvm/relay/testing/vgg.py) IR, and here is a list of functions that need to be written to have a fully compiled VGG net:

### Operations implementations for VGG Net
- [X] 1D, 2D Binary operations (divide, add, multiply, subtract) (#272, #275)
- [ ] 3D Binary operations
- [X] [nn.batch_matmul](https://tvm.apache.org/docs/api/python/relay/nn.html#tvm.relay.nn.batch_matmul) (#275)
- [X] [nn.relu](https://tvm.apache.org/docs/api/python/relay/nn.html#tvm.relay.nn.relu) (#275)
- [X] [nn.bias_add](https://tvm.apache.org/docs/api/python/relay/nn.html#tvm.relay.nn.bias_add) (#275)
- [X] [nn.batch_flatten](https://tvm.apache.org/docs/api/python/relay/nn.html#tvm.relay.nn.batch_flatten)(#272)
- [ ] [nn.conv2d](https://tvm.apache.org/docs/api/python/relay/nn.html#tvm.relay.nn.conv2d)
- [ ] [nn.batch_norm](https://tvm.apache.org/docs/api/python/relay/nn.html#tvm.relay.nn.batch_norm)
- [ ] [nn.max_pool2d](https://tvm.apache.org/docs/api/python/relay/nn.html#tvm.relay.nn.max_pool2d)
- [ ] [nn.dense](https://tvm.apache.org/docs/api/python/relay/nn.html#tvm.relay.nn.dense)
- [ ] [nn.dropout](https://tvm.apache.org/docs/api/python/relay/nn.html#tvm.relay.nn.dropout)
- [ ] [nn.softmax](https://tvm.apache.org/docs/api/python/relay/nn.html#tvm.relay.nn.softmax)
- [ ] [expand_dims](https://tvm.apache.org/docs/api/python/relay/index.html#tvm.relay.expand_dims)
- [ ] [sqrt](https://tvm.apache.org/docs/api/python/relay/index.html#tvm.relay.sqrt)
- [ ] [negative](https://tvm.apache.org/docs/api/python/relay/index.html#tvm.relay.negative)
- [ ] (All) How to take into account other function parameters, i.e. we won’t always want default kernel_size or axis.

---

### Type requirements
We also need to look into types. For example, I’ve implemented `ReLU`. However, this is meaningless when the input’s type is a tensor of unsigned integers.
- [ ] Support signed arrays for both integer and fixed point in Dahlia; currently trying to lower either returns an error.
- [ ] Support for signed operations in the primitives library for both integer and fixed point.
- [ ] Have output of `.dat` align with type rather than default to unsigned integers. For example, it'd be convenient to see signed numbers / fixed point numbers when they’re used rather than having to convert on paper from unsigned integers to the respective type when reading the output.
- [ ] Support 4-dimensional memory arrays. As discussed in previous sync-up, look into generalizing memory size by simply using row-major lookup on a 1-dimensional memory. If time-bound, this can be as simple as adding a 4-dimensional memory array to the primitives library in the same manner as the other dimensions.

---
### Clean-up
- [ ] Place the Dahlia functions into separate files, with cleaner templates. Sam suggested [Jinja 2](https://jinja.palletsprojects.com/en/2.11.x/intro/).

cgyurgyik, 2020-11-13:
As requested @yoonachang .
Currently, `decl A: bit<32>[8];` produces:
```
[Error] assertion failed: fuselang.common.CompilerError$NotImplemented: Arrays of signed integers. This feature is not yet implemented. Please open a feature request for it.
```
---
Similar for `decl A: fix<32, 32>[8];` :
```
[Error] assertion failed: fuselang.common.CompilerError$NotImplemented: Arrays of signed fixedpoints. This feature is not yet implemented. Please open a feature request for it.
```

That one statement should be enough to produce the respective error.

cgyurgyik, 2020-11-21:
**Current blocker:**
When lowering a function such as this one: `let %x1: Tensor[(1, 64), float32] = nn.bias_add(%x, %bias, axis=1);`
I cannot figure out a way to actually pull the `axis=1` argument. With `bias_add` we could infer based on the types of `x` and `bias`, but that's not always going to be the case. For example, in `nn.conv2d`, we may want to change the kernel size.

In the Python compiler, when visiting the call, it only gives us the first two arguments:
```
         free_var %x: Tensor[(1, 64), float32]
         free_var %bias: Tensor[(64), float32]
         nn.bias_add(%x, %bias) /* ty=Tensor[(1, 64), float32] */

```

sampsyo, 2020-11-21:
Neat! I'm glad you were able to track that down.

cgyurgyik, 2021-01-23:
Continued in #359.

closed 2021-01-23 by cgyurgyik