calyxir/calyx #282: D4 memory primitive "index out of bounds" error

cgyurgyik, 2020-11-23:
In #281 and [Dahlia #331](https://github.com/cucapra/dahlia/pull/331), I've added 4d memory primitives.
However, when I was trying to do a simple 4d memory add and lower it to verilog, I was getting the following error:
```
Stage 'futil' had a non-zero exit code.
--------------- stderr ---------------
thread 'main' panicked at 'index out of bounds: the len is 0 but the index is 0', .../futil/calyx/src/backend/traits.rs:29:27
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
---
Here's a simpler program that results in the same **error**:
```
decl X_BAD: ubit<32>[1][64][224][224];
let i = X_BAD[0][0][0][0];
```

Lowered to FuTIL:
```
import "primitives/std.lib";
component main() -> () {
  cells {
    X_BAD0_0_0_0 = prim std_mem_d4_ext(32,8,16,32,64,4,5,6,7);
    const0 = prim std_const(4,0);
    const1 = prim std_const(5,0);
    const2 = prim std_const(6,0);
    const3 = prim std_const(7,0);
    i_0 = prim std_reg(32);
  }
  wires {
    group upd0<"static"=1> {
      i_0.write_en = 1'd1;
      X_BAD0_0_0_0.addr3 = const3.out;
      X_BAD0_0_0_0.addr2 = const2.out;
      X_BAD0_0_0_0.addr1 = const1.out;
      X_BAD0_0_0_0.addr0 = const0.out;
      i_0.in = 1'd1 ? X_BAD0_0_0_0.read_data;
      upd0[done] = i_0.done ? 1'd1;
    }
  }
  control {
    upd0;
  }
}
```

---
However, the following lowers **without any issues**:
```
decl X_GOOD: ubit<32>[8][16][32];
let i = X_GOOD[0][0][0];
```
Lowered to FuTIL:
```
import "primitives/std.lib";
component main() -> () {
  cells {
    X_GOOD0_0_0 = prim std_mem_d3_ext(32,8,16,32,4,5,6);
    const0 = prim std_const(4,0);
    const1 = prim std_const(5,0);
    const2 = prim std_const(6,0);
    i_0 = prim std_reg(32);
  }
  wires {
    group upd0<"static"=1> {
      i_0.write_en = 1'd1;
      X_GOOD0_0_0.addr2 = const2.out;
      X_GOOD0_0_0.addr1 = const1.out;
      X_GOOD0_0_0.addr0 = const0.out;
      i_0.in = 1'd1 ? X_GOOD0_0_0.read_data;
      upd0[done] = i_0.done ? 1'd1;
    }
  }
  control {
    upd0;
  }
}
```

sampsyo, 2020-11-24:
Super interesting; thanks for the detailed report, @cgyurgyik! For complete context, would you mind doing the `RUST_BACKTRACE=1` thing to make Rust dump out a message that will pinpoint where the error is coming from?

cgyurgyik, 2020-11-24:
> Super interesting; thanks for the detailed report, @cgyurgyik! For complete context, would you mind doing the `RUST_BACKTRACE=1` thing to make Rust dump out a message that will pinpoint where the error is coming from?

**RUST_BACKTRACE=1**
```
thread 'main' panicked at 'index out of bounds: the len is 0 but the index is 0', /Users/cgyurgyik/Projects/futil-clone/futil/calyx/src/backend/traits.rs:29:27
stack backtrace:
   0: backtrace::backtrace::libunwind::trace
             at /Users/runner/.cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/libunwind.rs:86
   1: backtrace::backtrace::trace_unsynchronized
             at /Users/runner/.cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/mod.rs:66
   2: std::sys_common::backtrace::_print_fmt
             at src/libstd/sys_common/backtrace.rs:78
   3: <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt
             at src/libstd/sys_common/backtrace.rs:59
   4: core::fmt::write
             at src/libcore/fmt/mod.rs:1076
   5: std::io::Write::write_fmt
             at src/libstd/io/mod.rs:1537
   6: std::sys_common::backtrace::_print
             at src/libstd/sys_common/backtrace.rs:62
   7: std::sys_common::backtrace::print
             at src/libstd/sys_common/backtrace.rs:49
   8: std::panicking::default_hook::{{closure}}
             at src/libstd/panicking.rs:198
   9: std::panicking::default_hook
             at src/libstd/panicking.rs:217
  10: std::panicking::rust_panic_with_hook
             at src/libstd/panicking.rs:526
  11: rust_begin_unwind
             at src/libstd/panicking.rs:437
  12: core::panicking::panic_fmt
             at src/libcore/panicking.rs:85
  13: core::panicking::panic_bounds_check
             at src/libcore/panicking.rs:62
  14: <usize as core::slice::SliceIndex<[T]>>::index
             at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/libcore/slice/mod.rs:3103
  15: core::slice::<impl core::ops::index::Index<I> for [T]>::index
             at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/libcore/slice/mod.rs:2955
  16: <alloc::vec::Vec<T> as core::ops::index::Index<I>>::index
             at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/liballoc/vec.rs:1975
  17: calyx::backend::traits::Backend::run::{{closure}}
             at ./calyx/src/backend/traits.rs:29
  18: core::iter::adapters::map_fold::{{closure}}
             at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/libcore/iter/adapters/mod.rs:833
  19: core::iter::traits::iterator::Iterator::fold
             at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/libcore/iter/traits/iterator.rs:2022
  20: <core::iter::adapters::Map<I,F> as core::iter::traits::iterator::Iterator>::fold
             at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/libcore/iter/adapters/mod.rs:873
  21: core::iter::traits::iterator::Iterator::for_each
             at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/libcore/iter/traits/iterator.rs:658
  22: <alloc::vec::Vec<T> as alloc::vec::SpecExtend<T,I>>::spec_extend
             at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/liballoc/vec.rs:2140
  23: <alloc::vec::Vec<T> as alloc::vec::SpecExtend<T,I>>::from_iter
             at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/liballoc/vec.rs:2120
  24: <alloc::vec::Vec<T> as core::iter::traits::collect::FromIterator<T>>::from_iter
             at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/liballoc/vec.rs:1995
  25: core::iter::traits::iterator::Iterator::collect
             at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/libcore/iter/traits/iterator.rs:1671
  26: calyx::backend::traits::Backend::run
             at ./calyx/src/backend/traits.rs:26
  27: futil::cmdline::Opts::run_backend
             at src/cmdline.rs:132
  28: futil::main
             at src/main.rs:157
  29: std::rt::lang_start::{{closure}}
             at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/libstd/rt.rs:67
  30: std::rt::lang_start_internal::{{closure}}
             at src/libstd/rt.rs:52
  31: std::panicking::try::do_call
             at src/libstd/panicking.rs:348
[... 64 lines cut ...]
                               at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/liballoc/vec.rs:1995
  25:        0x101294e47 - core::iter::traits::iterator::Iterator::collect::h4d8b19fa9a3c53d0
                               at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/libcore/iter/traits/iterator.rs:1671
  26:        0x1012ec18f - calyx::backend::traits::Backend::run::hce2020d1c417b1ba
                               at /Users/cgyurgyik/Projects/futil-clone/futil/calyx/src/backend/traits.rs:26
  27:        0x10128f470 - futil::cmdline::Opts::run_backend::h92f38d2a50f6babb
                               at src/cmdline.rs:132
  28:        0x1012ae08e - futil::main::h5963969f4e008055
                               at src/main.rs:157
[... cut ...]
                               at src/libstd/panicking.rs:325
  33:        0x1016e4e8f - std::panic::catch_unwind::h5937a18f6925fee9
                               at src/libstd/panic.rs:394
  34:        0x1016e4e8f - std::rt::lang_start_internal::hcd84a36052901671
                               at src/libstd/rt.rs:51
  35:        0x10129f6e1 - std::rt::lang_start::hb65e82969f831813
                               at /Users/cgyurgyik/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/src/libstd/rt.rs:67
  36:        0x1012ae402 - main
```

sampsyo, 2020-11-25:
OK—I'm not, like, entirely sure here, but it looks like `calyx/src/backend/traits.rs:29` might be at least worth taking a glance at to see if it does any indexing that might be going out of bounds?

rachitnigam, 2020-11-25:
With https://github.com/cucapra/futil/commit/f13ba248f418b95216a1077243848fea515e8c92, you get a better error message:
```
Error: Mising verilog implementation for `std_mem_d4_ext`
```

This is because the `remove_external_memories` pass does not transform `std_mem_d4_ext` to `std_mem_d4`. Add a mapping and you'll be good to go.

cgyurgyik, 2020-11-27:
Thanks Adrian and Rachit!

PR #281 (2020-12-02): Add 4d memory primitive, dense, negative, expand_dims, broadcasting, attributes.
- Add 4d memory to primitive library; this is the highest dimension memory necessary for the VGG net, so should work for now. 
- Implements `negative` (won't work until signed arrays are supported), `expand_dims`.
- Implements `dense`, however, still need to add supports for `units` attribute.
- Add TVM attributes to the Dahlia declarations. This allows them to be accessed in the following manner:
```
// Example TVM IR: 
let %x1: Tensor[(1, 64), float32] = nn.bias_add(%x, %bias, axis=1);

...

// To access `axis`: [... cut]
closed 2020-12-02 by cgyurgyik