calyxir/calyx #284: Repeating output when looping using the same primitive.

cgyurgyik, 2020-11-29:
I'm trying to get down to the reason behind something I've discovered when trying to implement `nn.softmax`, and mentioned in #277. 

I've defined a `std_exp` component as follows:
```
primitive std_exp(exponent: 32, go: 1, clk: 1) -> (out: 32, done: 1) {
   verilog {
     module std_exp
       (input  logic [31:0]  exponent,
        input  logic        go,
        input  logic        clk,
        output logic [31:0] out,
        output logic        done);
        always_ff @(posedge clk) begin
          if (go) begin
            /* verilator lint_off REALCVT */
            out = 2.718281 ** exponent;
            done = 1;
          end
        end
     endmodule
  }
}
```

## Hard-coded Indices
The following Dahlia program **correctly** updates a `1 x 2` memory primitive:
```
import "std_exp.h" { def exp(x: ubit<32>): ubit<32>; }

decl A: ubit<32>[1][2];
decl Res: ubit<32>[1][2];

Res[0][0] := exp(A[0][0]);
---
Res[0][1] := exp(A[0][1]);
```

## Looping Through Indices
Now let's try and do that with loops. Here's the Dahlia program:
```
import "std_exp.h" { def exp(x: ubit<32>): ubit<32>; }

decl A: ubit<32>[1][2];
decl Res: ubit<32>[1][2];

for (let j: ubit<2> = 0..2) { Res[0][j] := exp(A[0][j]); }
```

This **doesn't** work. It just repeats the value at `Res[0][0]` for `Res[0][1], ... , Res[0][N]` (something that seems to occur in #277 as well).

Here's the FuTIL lowered from the Dahlia looping:
```
import "primitives/std.lib";
component main() -> () {
  cells {
    A0_0 = prim std_mem_d2_ext(32,1,2,1,2);
    A_read0_0 = prim std_reg(32);
    Res0_0 = prim std_mem_d2_ext(32,1,2,1,2);
    add0 = prim std_add(2);
    const0 = prim std_const(2,0);
    const1 = prim std_const(2,1);
    const2 = prim std_const(1,0);
    const3 = prim std_const(1,0);
    const4 = prim std_const(2,1);
    exp0 = prim std_exp();
    j0 = prim std_reg(2);
    le0 = prim std_le(2);
  }
  wires {
    group cond0<"static"=0> {
      cond0[done] = 1'd1;
      le0.left = j0.out;
      le0.right = const1.out;
    }
    group let0<"static"=1> {
      j0.in = const0.out;
      j0.write_en = 1'd1;
      let0[done] = j0.done;
    }
    group upd0<"static"=1> {
      A_read0_0.write_en = 1'd1;
      A0_0.addr1 = j0.out;
      A0_0.addr0 = const2.out;
      A_read0_0.in = 1'd1 ? A0_0.read_data;
      upd0[done] = A_read0_0.done ? 1'd1;
    }
    group upd1 {
      Res0_0.addr1 = j0.out;
      Res0_0.addr0 = const3.out;
      Res0_0.write_en = exp0.done;
      exp0.exponent = A_read0_0.out;
      exp0.go = !exp0.done ? 1'd1;
      Res0_0.write_data = exp0.done ? exp0.out;
      upd1[done] = Res0_0.done ? 1'd1;
    }
    group upd2<"static"=1> {
      j0.write_en = 1'd1;
      add0.left = j0.out;
      add0.right = const4.out;
      j0.in = 1'd1 ? add0.out;
      upd2[done] = j0.done ? 1'd1;
    }
  }
  control {
    seq {
      let0;
      while le0.out with cond0 {
        seq {
          upd0;
          upd1;
          upd2;
        }
      }
    }
  }
}
```

## Adding a second `std_exp` primitive.
--- 
I then decided to try the following: Instead of looping twice (once for each value in the array), just copy `upd0`, `upd1`, and `upd2`, have the control be: `seq { upd0; upd1; upd2; upd3; upd4; upd5 }`. This still didn't work. I then took one additional step: create a second primitive `exp1 = prim std_exp()`. This worked! Having a separate `std_exp()` to update each memory value worked. Below is the FuTIL for the working version:
```
import "primitives/std.lib";
component main() -> () {
  cells {
    A0_0 = prim std_mem_d2_ext(32,1,2,1,2);
    A_read0_0 = prim std_reg(32);
    Res0_0 = prim std_mem_d2_ext(32,1,2,1,2);
    add0 = prim std_add(2);
    const0 = prim std_const(2,0);
    const1 = prim std_const(2,1);
    const2 = prim std_const(1,0);
    const3 = prim std_const(1,0);
    const4 = prim std_const(2,1);
    exp0 = prim std_exp();
    exp1 = prim std_exp();
    j0 = prim std_reg(2);
    le0 = prim std_le(2);
  }
  wires {
    group let0<"static"=1> {
      j0.in = const0.out;
      j0.write_en = 1'd1;
      let0[done] = j0.done;
    }
    group upd0<"static"=1> {
      A_read0_0.write_en = 1'd1;
      A0_0.addr1 = j0.out;
      A0_0.addr0 = const2.out;
      A_read0_0.in = 1'd1 ? A0_0.read_data;
      upd0[done] = A_read0_0.done ? 1'd1;
    }
    group upd1 {
      Res0_0.addr1 = j0.out;
      Res0_0.addr0 = const3.out;
      Res0_0.write_en = exp0.done;
      exp0.exponent = A_read0_0.out;
      exp0.go = !exp0.done ? 1'd1;
      Res0_0.write_data = exp0.done ? exp0.out;
      upd1[done] = Res0_0.done ? 1'd1;
    }
    group upd2<"static"=1> {
      j0.write_en = 1'd1;
      add0.left = j0.out;
      add0.right = const4.out;
      j0.in = 1'd1 ? add0.out;
      upd2[done] = j0.done ? 1'd1;
    }
    group upd3<"static"=1> {
      A_read0_0.write_en = 1'd1;
      A0_0.addr1 = j0.out;
      A0_0.addr0 = const2.out;
      A_read0_0.in = 1'd1 ? A0_0.read_data;
      upd3[done] = A_read0_0.done ? 1'd1;
    }
    group upd4 {
      Res0_0.addr1 = j0.out;
      Res0_0.addr0 = const3.out;
      Res0_0.write_en = exp1.done;
      exp1.exponent = A_read0_0.out;
      exp1.go = !exp1.done ? 1'd1;
      Res0_0.write_data = exp1.done ? exp1.out;
      upd4[done] = Res0_0.done ? 1'd1;
    }
    group upd5<"static"=1> {
      j0.write_en = 1'd1;
      add0.left = j0.out;
      add0.right = const4.out;
      j0.in = 1'd1 ? add0.out;
      upd5[done] = j0.done ? 1'd1;
    }
  }
  control {
    seq {
      let0;
      upd0;
      upd1;
      upd2;
      upd3;
      upd4;
      upd5;
    }
  }
}
```

## Remarks
- Right now, I don't have anything mapped for the static timing of `std_exp`.
- I'm trying to provide a simpler example that doesn't use `std_exp`, but haven't been able to do so yet.

rachitnigam, 2020-11-29:
Every variable inside an `always_ff` block gets turned into a register or a latch. Your code, which only assigns to done in the true branch creates a latch. This means that even when the `go` signal becomes `0`, the done signal is kept at `1` (the value in the done register is "latched"). This is obvious from the waveform:

[image]

[Latches are bad](https://www.nandland.com/articles/how-to-avoid-transparent-latches-in-vhdl-and-verlog.html). You very rarely want to implement a latch. As a rule of thumb, every conditional construct (if, switch) should have default assignments to every variable

There are two fixes:
1. Assign to every variable in an else condition:
```
primitive std_exp(exponent: 32, go: 1, clk: 1) -> (out: 32, done: 1) {
   verilog {
     module std_exp
       (input  logic [31:0]  exponent,
        input  logic        go,
        input  logic        clk,
        output logic [31:0] out,
        output logic        done);
        always_ff @(posedge clk) begin
          if (go) begin
            /* verilator lint_off REALCVT */
            out = 2.718281 ** exponent;
            done = 1;
          end else begin
            out = 0;
            done = 0;
          end
        end
     endmodule
  }
}
```
2. Implement combinational exponent:
```
primitive std_exp(exponent: 32, go: 1, clk: 1) -> (out: 32, done: 1) {
   verilog {
     module std_exp
       (input  logic [31:0]  exponent,
        output logic [31:0] out);
        /* verilator lint_off REALCVT */
        assign out = 2.718281 ** exponent;
        end
     endmodule
  }
}
```
A combinational exponent circuit is not going to be synthesizable (i.e. it can't be mapped onto an FPGA).

cgyurgyik, 2020-11-29:
Thanks, looking at the waveform should've been an obvious next step. I initially tried doing something similar to option 1 since that's essentially what the `std_sqrt` operator uses, however if I recall correctly, I couldn't get it to work. I'll look more into this later.

rachitnigam, 2020-11-29:
Using 1 generates the right waveform outputs (20 and 7):

[image]

cgyurgyik, 2020-11-30:
Can you show me the FuTIL code? I'm not getting that. When using 1 and a loop, `out[31:0]` is 0 across the board.

rachitnigam, 2020-11-30:
It’s the code you pasted above. I just changed the primitive definition. It could be that there are other problems but it seems that the primitive outputs the right results.

cgyurgyik, 2020-11-30:
Just setting the `out` value in the `else` case doesn't work when `done = 0`. However, this did work:
```
primitive std_exp(exponent: 32, go: 1, clk: 1) -> (out: 32, done: 1) {
   verilog {
     module std_exp
       (input  logic [31:0]  exponent,
        input  logic        go,
        input  logic        clk,
        output logic [31:0] out,
        output logic        done);
        always_ff @(posedge clk) begin
          if (go) begin
            /* verilator lint_off REALCVT */
            out = 2.718281 ** exponent;
            done = 1;
          end else begin
            out = 2.718281 ** exponent;
            done = 1;
          end
        end
     endmodule
  }
}
```

I then changed it from `Res[i][j] := exp(A[i][j]);` -> `Res[i][j] := exp(A[i][j]) / 2;` and it results in zeroes again for `Res[i][j]`. I'm going to take a look at the waveform later.

rachitnigam, 2020-11-30:
This implementation is incorrect. If you look at the logic generated in the group, the `go` signal is only turned off when `done` is `0`.

You've, in essence, re-implement option (2) which combinationally (continuously) outputs the same value on the output port based on the exponent. When you try to implement a pipelined exponent, this will fail.

cgyurgyik, 2020-11-30:
Heh, yeah that makes sense. So going back to the following:
```
 ...
 end else begin
   out = out;
   done = 0;
```

It seems that the waveform output below looks correct for `std_exp`. However, the `Res_write_data` is not getting set to `1`, and the `Res_read_data` remains `0`? I'll need to take a look at the FuTIL code again for this.

[image]

I believe we'd like something like this:
[image]

rachitnigam, 2020-11-30:
No, it's being set to 1. There is a transition for `res_write_data` that overlaps with the `write_en` signal. I'm not sure what's going on. Might require deeper investigation.

cgyurgyik, 2020-11-30:
Ah, zooming in helps. Here are the actual values for above (well the first half at least):

[image]

rachitnigam, 2020-11-30:
Looking at the `mem` component inside `Res0_0` is what caused me to be suspicious. It's never being updated for some reason...

cgyurgyik, 2020-11-30:
> Looking at the `mem` component inside `Res0_0` is what caused me to be suspicious. It's never being updated for some reason...

If I'm not mistaken, we should expect that `Res0_0_read_data == 20` when `write_en` is low again? However, right now, no transition is occurring for `Res0_0_read_data`.

rachitnigam, 2020-12-01:
Yup. Confusingly, if you add a fake register to std_mem_d2, it starts working as expected. I haven’t got the slightest clue why

rachitnigam, 2020-12-01:
This is the patch:
```diff
From 909fa64718db9827da5b2795776ee5e37db3844c Mon Sep 17 00:00:00 2001
From: Rachit Nigam <rachit.nigam12@gmail.com>
Date: Mon, 30 Nov 2020 19:45:34 -0500
Subject: [PATCH] fake changes

---
 primitives/std.lib | 24 ++++++++++++++++++++++++
 1 file changed, 24 insertions(+)

diff --git a/primitives/std.lib b/primitives/std.lib
index 805384c..0428c22 100644
--- a/primitives/std.lib
+++ b/primitives/std.lib
@@ -71,11 +71,13 @@ primitive std_mem_d2[width, d0_size, d1_size, d0_idx_size, d1_idx_size](
 
       /* verilator lint_off WIDTH */
       logic [width-1:0]  mem[d0_size-1:0][d1_size-1:0];
+      logic [width-1:0]  fake;
 
       assign read_data = mem[addr0][addr1];
       always_ff @(posedge clk) begin
         if (write_en) begin
           mem[addr0][addr1] <= write_data;
+          fake <= write_data;
           done <= 1'd1;
         end else
           done <= 1'd0;
@@ -878,3 +880,25 @@ primitive fixed_p_std_add_dbit[width, int_width1, fract_width1, int_width2, frac
       endmodule
     }
 }
+
+primitive std_exp(exponent: 32, go: 1, clk: 1) -> (out: 32, done: 1) {
+   verilog {
+     module std_exp
+       (input  logic [31:0]  exponent,
+        input  logic        go,
+        input  logic        clk,
+        output logic [31:0] out,
+        output logic        done);
+        always_ff @(posedge clk) begin
+          if (go) begin
+            /* verilator lint_off REALCVT */
+            out = 2.718281 ** exponent;
+            done = 1;
+          end else begin
+            done = 0;
+            out = 0;
+          end
+        end
+     endmodule
+  }
+}
-- 
2.21.1 (Apple Git-122.3)

```

There is a `fake` register in std_mem_d2 that is sits there and reads the value from `write_data`. My best guess is that there is a verilator bug somewhere that is causing the signals to `write_data` being dropped somehow.

cgyurgyik, 2020-12-01:
[image]

I added a register `RegisterMcRegisterFace` and wrote the output of `exp` to it, and had `Res0_0` read from it. That worked.

rachitnigam, 2020-12-01:
Almost certain that this is some weird verilator optimization bug. Instead of adding a fake signal, adding the following comment to the `std_mem_d2` definition also fixes it:
```
        input logic [width-1:0]   write_data /*verilator public*/,
```
I assume making the name public forces verilator to not optimize it.

cgyurgyik, 2020-12-01:
Hmm trying to add an additional division however goes back to outputting zeroes.
E.g.
```for (let j: ubit<2> = 0..2) { Res[0][j] := exp(A[0][j]) / 1; }```

[image]

Multiplication seems to work fine though.

rachitnigam, 2020-12-01:
Is that with the last solution I suggested?

cgyurgyik, 2020-12-01:
> Is that with the last solution I suggested?

With `/*verilator public*/`, yes.

rachitnigam, 2020-12-01:
Ugh, there is something deeper going on here.

rachitnigam, 2020-12-01:
Tracking issue in Verilator: https://github.com/verilator/verilator/issues/2661

rachitnigam, 2020-12-02:
@cgyurgyik can you try the fixes on #288 and see if they help?

cgyurgyik, 2020-12-02:
With #288:

Still works:
`for (let j: ubit<2> = 0..2) { Res[0][j] := exp(A[0][j]); }`

Still does **not** work:
`for (let j: ubit<2> = 0..2) { Res[0][j] := exp(A[0][j]) / 1; }` 

[image]

rachitnigam, 2020-12-02:
Can you paste the FuTIL code for the second case?

cgyurgyik, 2020-12-02:
I simplified it to this, as it may be unrelated:
```
import "std_exp.h" {
  def exp(x: ubit<32>): ubit<32>;
}

decl A: ubit<32>[1];
decl Res: ubit<32>[1];

Res[0] := exp(A[0]) / 1 ; // Produces 0, unless we remove division by 1.
```

FuTIL
```
import "primitives/std.lib";
component main() -> () {
  cells {
    A0 = prim std_mem_d1_ext(32,1,1);
    A_read0_0 = prim std_reg(32);
    Res0 = prim std_mem_d1_ext(32,1,1);
    bin_read0_0 = prim std_reg(32);
    const0 = prim std_const(1,0);
    const1 = prim std_const(32,1);
    const2 = prim std_const(1,0);
    div_pipe0 = prim std_div_pipe(32);
    exp0 = prim std_exp();
  }
  wires {
    group let0 {
      bin_read0_0.in = div_pipe0.out;
      bin_read0_0.write_en = div_pipe0.done;
      let0[done] = bin_read0_0.done;
      div_pipe0.left = exp0.out;
      div_pipe0.right = const1.out;
      div_pipe0.go = !div_pipe0.done ? 1'd1;
      exp0.exponent = A_read0_0.out;
      exp0.go = !exp0.done ? 1'd1;
    }
    group upd0<"static"=1> {
      A_read0_0.write_en = 1'd1;
      A0.addr0 = const0.out;
      A_read0_0.in = 1'd1 ? A0.read_data;
      upd0[done] = A_read0_0.done ? 1'd1;
    }
    group upd1<"static"=1> {
      Res0.addr0 = const2.out;
      Res0.write_en = 1'd1;
      Res0.write_data = 1'd1 ? bin_read0_0.out;
      upd1[done] = Res0.done ? 1'd1;
    }
  }
  control {
    seq {
      upd0;
      let0;
      upd1;
    }
  }
}
```

cgyurgyik, 2020-12-02:
Just quickly trying, this does work:
```
decl A: ubit<32>[1];
decl Res: ubit<32>[1];

Res[0] := exp(A[0]);
---
Res[0] := Res[0] / 1;
```

```
import "primitives/std.lib";
component main() -> () {
  cells {
    A0 = prim std_mem_d1_ext(32,1,1);
    A_read0_0 = prim std_reg(32);
    Res0 = prim std_mem_d1_ext(32,1,1);
    Res_read0_0 = prim std_reg(32);
    bin_read0_0 = prim std_reg(32);
    const0 = prim std_const(1,0);
    const1 = prim std_const(1,0);
    const2 = prim std_const(1,0);
    const3 = prim std_const(32,1);
    const4 = prim std_const(1,0);
    div_pipe0 = prim std_div_pipe(32);
    exp0 = prim std_exp();
  }
  wires {
    group let0 {
      bin_read0_0.in = div_pipe0.out;
      bin_read0_0.write_en = div_pipe0.done;
      let0[done] = bin_read0_0.done;
      div_pipe0.left = Res_read0_0.out;
      div_pipe0.right = const3.out;
      div_pipe0.go = !div_pipe0.done ? 1'd1;
    }
    group upd0<"static"=1> {
      A_read0_0.write_en = 1'd1;
      A0.addr0 = const0.out;
      A_read0_0.in = 1'd1 ? A0.read_data;
      upd0[done] = A_read0_0.done ? 1'd1;
    }
    group upd1<"static"=2> {
      Res0.addr0 = const1.out;
      Res0.write_en = exp0.done;
      exp0.exponent = A_read0_0.out;
      exp0.go = !exp0.done ? 1'd1;
      Res0.write_data = exp0.done ? exp0.out;
      upd1[done] = Res0.done ? 1'd1;
    }
    group upd2<"static"=1> {
      Res_read0_0.write_en = 1'd1;
      Res0.addr0 = const2.out;
      Res_read0_0.in = 1'd1 ? Res0.read_data;
      upd2[done] = Res_read0_0.done ? 1'd1;
    }
    group upd3<"static"=1> {
      Res0.addr0 = const4.out;
      Res0.write_en = 1'd1;
      Res0.write_data = 1'd1 ? bin_read0_0.out;
      upd3[done] = Res0.done ? 1'd1;
    }
  }
  control {
    seq {
      upd0;
      upd1;
      upd2;
      let0;
      upd3;
    }
  }
}
```

rachitnigam, 2020-12-02:
Yup, that makes sense. This is the problem mentioned in #264. Take a look at this group:
```
    group let0 {
      exp0.exponent = A_read0_0.out;
      exp0.go = !exp0.done ? 1'd1;

      div_pipe0.left = exp0.out; // ASSUMES exp0.out is done immediately
      div_pipe0.right = const1.out;
      div_pipe0.go = !div_pipe0.done ? 1'd1;

      bin_read0_0.in = div_pipe0.out;
      bin_read0_0.write_en = div_pipe0.done;

      let0[done] = bin_read0_0.done;
    }
```

The short term fix is to have the backend rewrite:
```
let x = exp(Res[0]) / 1;
```
into:
```
let _tmp = expr(Res[0]);
---
let x = _tmp / 1;
```

Open an issue in the Dahlia repo saying that function calls need to be stored in temporaries.

rachitnigam, 2020-12-02:
@cgyurgyik can you close this issue when you confirm that the suggested solution works?

PR #288 (2020-12-02): Fixes to avoid zero-time reset bug in simulation
Fixes #284.

Lack of initial inputs causes a zero-time reset bug in simulation. See verilator/verilator#2661.
closed 2020-12-02 by rachitnigam