calyxir/calyx #296: minimize_regs incorrectly handles writes in par blocks that escape component boundaries

rachitnigam, 2020-12-06:
This program mis-compiles:
```
import "primitives/std.lib";
component main() -> (x_out: 32) {
  cells {
    x_0 = prim std_reg(32);
    y_0 = prim std_reg(32);
  }
  wires {
    group let0<"static"=1> {
      x_0.in = 32'd1;
      x_0.write_en = 1'd1;
      let0[done] = x_0.done;
    }
    group let1<"static"=1> {
      y_0.in = 32'd2;
      y_0.write_en = 1'd1;
      let1[done] = y_0.done;
    }
  }
  control {
    par {
      let0;
      let1;
    }
  }
}

```
compiles into:
```
import "primitives/std.lib";
component main(go: 1, clk: 1) -> (x_out: 32, done: 1) {
  cells {
    x_0 = prim std_reg(32);
    y_0 = prim std_reg(32);
  }
  wires {
    group let0<"static"=1> {
      x_0.in = 32'd1;
      x_0.write_en = 1'd1;
      let0[done] = x_0.done;
    }
    group let1<"static"=1> {
      x_0.in = 32'd2;
      x_0.write_en = 1'd1;
      let1[done] = x_0.done;
    }
  }

  control {
    par {
      let0;
      let1;
    }
  }
}
```

The input fuse code is:
```
let x: ubit<32> = 1;
let y: ubit<32> = 2;
```

sgpthomas, 2020-12-09:
`x_out` is never written to in this program. so the rewriting doesn't change the semantics of the component.

rachitnigam, 2020-12-09:
It changes the semantics of the program because the latter is an invalid program. Multiple drives to the same port aren’t allowed.

rachitnigam, 2020-12-09:
A more thoughtful answer: It is true that the conflicting read cannot be observed because the values never affect any external ports. However, that brings us to the more interesting question of what guarantees do we have for cells and assignments that are not externally visible.

For example, LLVM's spec dictates that every *reachable* basic block needs to be in SSA. This distinction is important. (@sampsyo might have thoughts on this)

Regardless, in this particular case, the assignment ends up creating multiple drives to the same port which violates hardware well-formedness constraint.

sgpthomas, 2020-12-09:
The bug is that `x0` and `y0` are not live going into the groups or coming out of the groups (because both groups kill their respective registers and they aren't alive anywhere else). the thing we want is the variables alive inside of a group, rather than coming out of a group.

regarding the deeper question, I'll have to think more about it. but off the cuff it seems like the only constraints are that we maintain visible side effects and we maintain hardware well-formedness.

sampsyo, 2020-12-10:
This is an interesting question. The question y'all have narrowed in on is not so much about the optimization but about the definition of well-formedness. Should it be:

1. No component is ever activated twice 'simultaneously,' i.e., by activations that are unordered by sequencing control.
2. Something more subtle than this, where the above kind of simultaneous activation _can_ happen but it is illegal to read out the state of any registers that may have been affected by this simultaneous activation.

I'm inclined to say that we should prefer something more like option 1 above than option 2 because the criterion is so much easier to understand. That is, it seems a bit worrisome to get into a situation where we're saying "you can do this seemingly-bad thing, and that's fine, but what you should _not_ do is read the result." For example, is it legal to read that bad register value and throw it away? Does it have to be that the value flows to an external port for the program to be illegal? It's sort of fuzzy and ends up being a slippery slope.

Would one possible remedy for this optimization be to delete _both_ registers because they are unused?

sampsyo, 2020-12-10:
(With a name like "minimize_regs," you would hope it would be able to delete _all_ the registers! 😆)

commit 03ed3511 (2020-12-18): add test for minimize regs (escaping boundaries) (closes #296)
closed 2020-12-18 by sgpthomas