calyxir/calyx #306: Fud always outputting zero

yoonachang, 2020-12-12:
To test the implementation in fud fixedpoint I wanted to sum elements in two arrays and out put in and resulting array.
I compiled 
This code wrriten in dahlia 
`decl a0_0: ubit<16>[2][2];
decl b0_0: ubit<16>[2][2];
decl result0_0: ubit<16>[2][2];

for (let i: ubit<2> = 0..2) {
  for (let j: ubit<2> = 0..2) {
    result0_0[i][j] := a0_0[i][j] + b0_0[i][j];
  }
}`

And tried to input 
`{
    "a0_00_0": {
        "data": [
          [3,12],
          [4,10]
        ],
        "type": "TSizedInt(16, true)"
      },
      "b0_00_0": {
        "data": [
          [23,12],
          [14,2]
        ],
        "type": "TSizedInt(16, true)"
      },
      "result0_00_0": {
        "data": [
          [0,0],
          [0,0]
        ],
        "type": "TSizedInt(16, true)"
      }
  }`
As data

Some how result is always 0 in both .dat file and in .vcd

rachitnigam, 2020-12-12:
Have you tried looking at the VCD file? If the VCD file doesn’t contain the right values for the memories, it’s not a Fud 
problem.

rachitnigam, 2020-12-12:
Are you sure that the memories should be named “a0_00_0” in the data file?

yoonachang, 2020-12-12:
For the name of the memories yeah, it used to be something else but it wouldn't run, that's why I changed it. I tried looking at the vcd file with scansion and there were all zeros

rachitnigam, 2020-12-12:
What branch are you working on? Can you commit exactly the code that you’re using? Also, one thing to do is check if the input memories have the right values.

yoonachang, 2020-12-12:
I'm at branch fxdpoint_dat I'm trying to test intmatadd.futil with `{
    "a0_00_0": {
        "data": [
          [3,12],
          [4,10]
        ],
        "type": "TSizedInt(16, true)"
      },
      "b0_00_0": {
        "data": [
          [23,12],
          [14,2]
        ],
        "type": "TSizedInt(16, true)"
      },
      "result0_00_0": {
        "data": [
          [0,0],
          [0,0]
        ],
        "type": "TSizedInt(16, true)"
      }
  }`

sampsyo, 2020-12-13:
Just one other idea for how to debug: try minimizing the program. Does it still output just zeroes if you delete all the code and replace it with a no-op? How about if you just have one loop, and not a doubly-nested loop? Exploring this space may help "blame" something in the input program for triggering the bug, which would in turn help narrow down the bug itself.

yoonachang, 2020-12-14:
I'm trying this out with a simpler code 
[image]

rachitnigam, 2020-12-28:
Notes from debugging with @yoonachang:

Seems like the design runs for one clock cycle on YooNa's machine. However, on my machine, this does seem to be the case. This bug, in theory, in very similar to #288. However, without me being able to reproduce it, I don't know what do to next.

One thing to try out is using a different simulator. It might be worth setting up multiple simulators just so we can catch simulation bugs.

sampsyo, 2020-12-30:
Ultra weird. Could be worth isolating where the bug arises: Verilator or the C++ compiler (or something else weird) by generating the C++ and sending it somewhere else to compile/execute. Or, for the time being, just trying to normalize by running on Gorgonzola…

rachitnigam, 2021-03-04:
AFAICT, this is non-actionable.

closed 2021-03-04 by rachitnigam