calyxir/calyx #311: Attributes everywhere

rachitnigam, 2020-12-16:
Attribute annotations are currently limited to just groups. This proposal makes them available to cells, ports, and components. Attributes should become a general mechanism to attach extra information to various parts of the IL.

As a starting example, consider a type recovery pass that walks over cells and ports and tries to come up with `signed` or `unsigned` type for each:

```
component foo(@type("signed") inp1: 32) -> (@type("signed") out: 32) {
  cells {
    @type("unsigned") r = std_reg(32);
  }
}
```

Following the principle laid out in #270 ("everything that is solved for groups must be solved for components"), we can attach `"static"` information to a component and change the compilation of `invoke` (#273) to generate a faster FSM.

sampsyo, 2020-12-16:
Nice. This is a cool idea. And I like the implicit proposal that we could use this for an optional "overlay" type system that actually represents data types.

sgpthomas, 2020-12-16:
would be a useful debugging tool as well. for an analysis like live-range-analysis, it would be super useful to be able to annotate the things live at each group

rachitnigam, 2020-12-30:
An annotation called `"fsm"` that the verilog backend can use to generate optimized `case`-based transition:
```
cells {
  @encoding("fsm") fsm = std_reg(32); // <- added during compilation
}
```

closed 2021-01-04 by rachitnigam

cgyurgyik, 2022-12-15:
Another idea is following [LLVM's opaque pointer approach](https://llvm.org/docs/OpaquePointers.html), i.e., give types to reads/writes instead of the underlying memory. This is similar to what our binary operators currently do, and would ensure we don't have type clashing for the memory, which will likely be reinterpreted many different ways.

```
component foo(@type("signed") inp1: 32) -> (@type("signed") out: 32) {
  cells {
    r = std_reg(32);
  }
  wires {
    @type("unsigned") out = r.out; 
  }
}
```

PR #340 (2021-01-04): Add attribute points
Add attribute points for components, cell declarations, and port definitions:
```
import "primitives/core.futil";
component main<"static"=1>(@stable(1) @go_port(1) in: 32) -> (@stable(0) out: 32) {
  cells {
    @precious(1) out = prim std_reg(32);
    le = prim std_le(32);
  }
  wires {
    group cond<"stable"=1> {
      cond[done] = 1'd1;
    }
  }
  control {
  }
}
```

Fixes #311.