calyxir/calyx #313: better error message when `vivado` and `vivado_hls` can't find the right files

sgpthomas, 2020-12-16:


rachitnigam, 2020-12-16:
Example?

rachitnigam, 2021-01-24:
@sgpthomas One more ping to explain this issue. Which files specifically?

sgpthomas, 2021-01-25:
Different versions of vivado produce different report files. The stages that take the synthesis files and produce json of the specific metrics assume that particular files exist. We should simply add a check to see if a file exists, and if it doesn't provide a more helpful error message.

One way to reproduce. On gorgonzola:
```
source /opt/Xilinx/SDx/2017.2/settings64.sh
cd ~/futil-evaluation/benchmarks/polybench # assuming this exists
fud e linear-algebra-2mm.fuse --to hls-estimate
```
this produces:
```
⠦ vivado-hls → hls-files[Errno 2] No such file or directory: '/tmp/tmpchjz7vl9/benchmark.prj/solution1/solution1_data.json'
HLS files weren't found, skipping.
Traceback (most recent call last):
  File "/home/sgt43/.local/bin/fud", line 8, in <module>
    sys.exit(main())
  File "/home/sgt43/.local/lib/python3.6/site-packages/fud/main.py", line 150, in main
    exec.run_fud(args, cfg)
  File "/home/sgt43/.local/lib/python3.6/site-packages/fud/exec.py", line 88, in run_fud
    last=i == (len(path) - 1)
  File "/home/sgt43/.local/lib/python3.6/site-packages/fud/stages/__init__.py", line 149, in transform
    res = step.run(prev_out, ctx=ctx, dry_run=dry_run, last=last)
  File "/home/sgt43/.local/lib/python3.6/site-packages/fud/stages/__init__.py", line 204, in run
    return self.func(input_src, ctx)
  File "/home/sgt43/.local/lib/python3.6/site-packages/fud/stages/__init__.py", line 281, in f
    return func(inp, ctx)
  File "/home/sgt43/.local/lib/python3.6/site-packages/fud/stages/vivado_hls.py", line 92, in f
    return (Source(BytesIO(res.encode("UTF-8")), SourceType.File), None, 0)
AttributeError: 'NoneType' object has no attribute 'encode'
```

commit a9ba3b9a (2021-01-25): better error for missing files for estimate stages (fixes #313)
closed 2021-01-25 by sgpthomas