calyxir/calyx #314: overly conservative flaw with `live_range_analysis` and `minimize_regs`

sgpthomas, 2020-12-16:
When different registers are used in parallel, we can't share them because we don't have an ordering over them and thus don't know if sharing them will change the semantics of the program.

To deal with this, we say that something is live under the normal dataflow rules + if it is used in parallel. This seemingly benign assumption actually causes us to miss most sharing opportunities with things running in parallel. consider:
```
par {
  seq { wr_x; ...; rd_x; wr_y; ...; rd_y } // THREAD A
  seq { wr_a; ...; rd_a; wr_b; ...; rd_b } // THREAD B
}
```
Assuming no other uses of `x` and `y`, we can rewrite thread `A` to use only one register. similarly for `B`.

However, if we apply the rule mentioned earlier to define liveness, here are the live variables that we get:
```
par {
  seq { 
    wr_x; ...; rd_x; // live x, a, b
    wr_y; ...; rd_y; // live y, a, b
  } 
  seq { 
    wr_a; ...; rd_a; // live a, x, y
    wr_b; ...; rd_b; // live b, x, y
  }
}
```
We perform sharing by constructing a conflict graph where there are edges between registers if they are every alive at the same time. Because of the way that we defined liveness (things are alive if used in parallel), we can't do any sharing here even though it's safe. `x` and `y` should be shareable, but they are alive together in thread B. Similarly for `a` and `b`.

I think the problem here is a conflation of two notions: liveness at a point, and used in parallel. If these notions are separate, then the conflict graph can be built so that the graph is built normally in each thread, and then all the registers used in parallel also conflict.

rachitnigam, 2020-12-16:
There is a similar problem in resource-sharing. The `all_conflicting` method solves this by adding the edges *after* traversing all of the children of a `par`.

sampsyo, 2020-12-16:
Yeah!!!! This is a good demonstration of the problem we chatted about a while back, Sam: CFG-like ordering edges and HB edges are of very different "flavors." I think a complete solution to this kind of thing inevitably has to treat those kinds of ordering differently… and something that reduces everything to the sequential case might inevitably be conservative. Interesting!

commit 3f0efcc4 (2021-01-07): fixed #314 by using ScheduleConflicts to construct edges between things used in parallel
closed 2021-01-07 by sgpthomas