calyxir/calyx #315: Generate `dat` files from FuTIL programs

sgpthomas, 2020-12-17:
It would be nice if something (probably fud? maybe the compiler) could look at a FuTIL program and generate a data json file with random data / zeroed out data. This would make it easier to go from writing a program -> testing it, makes it easier to give instructions to people about how to simulate FuTIL, etc.

rachitnigam, 2020-12-17:
Its not going to be quite as easy since relevant type information is lost when generating FuTIL programs (signed vs unsigned vs fixed point).

sampsyo, 2020-12-18:
This would indeed be super helpful… a few thoughts come to mind.

- We could ask the Dahlia compiler to generate this. Won't help with other frontends, obvi.
- The thing could just start by generating the structure with the names of memories, not the types. Could be better than nothing.
- It would certainly need help from the Calyx compiler, not just fud… because you need to parse the program at the very least, right? And you don't want to write a _second_ parser for the same language. :smiley:

It might also be worthwhile to think out of the box about other ways to make it easier to get from writing a Calyx program to running it, as @sgpthomas. Like better "linting" for JSON inputs, even if we don't actually generate them, for instance.

rachitnigam, 2021-06-07:
We've started marking memories with the `@external(1)` annotation when we want to read/write values from them during simulation. This means that all this imaginary tool needs to do is walk over the memories and generate a JSON file based on the primitive they use. In the future, if the memories have more attributes that describe the shape of the data (bit/float/fixed-point) and its signedness, we can also use that to generate more relevant data.

PR #1003 (2022-05-19): Data gen
Added my tool that generates the json from a .futil input. 

The file it writes into is hardcoded to "data.json". 

Edit: I don't think it worked the way it was supposed to. So I think a problem is that for some reason git thinks my tools/data_gen folder is its own github repo for some reason. When I'm on my local computer that's how it's treating it. 

I deleted the branch on my local computer and just retried everything, making sure to not have a .git file in my tools/data_gen. You can see everything, but still doesn't pass tests. 

Edit(rachit): Fixes #315.
closed 2022-05-19 by calebmkim