calyxir/calyx #323: FuTIL imports and "thin" libraries

rachitnigam, 2020-12-22:
FuTIL currently only allows `import` statements for FuTIL libraries. #268 tries to solve this by removing the distinction between libraries files (that can only contain `primitive` definitions) and FuTIL files (that can only contain `component` definitions). It allows arbitrary mixing of both kinds of definitions.

While we're at it, I think we should reconsider the idea of "fat" libraries where the primitive definitions embed the Verilog implementation inside them. Instead, we should use Dahlia style `import` statements that define the interface for primitives without implementing them.

A few reasons to do this:
1. Being able to put primitives inside `.sv` files and testing them separately from the compiler. Furthermore, in the future, we can try to automatically generate `primitive` interfaces from an `.sv` file.
2. We cannot manipulate the strings embedded in the library files. Why bother carrying them throughout the compilation process?

sampsyo, 2020-12-23:
Yeah, this is probably a good idea. FWIW, I think a good general term for what this would look like is `extern` declarations… i.e., declarations without definitions. Then the programs would need to be "linked" after the fact.

Thinking back to why we went with the "fat library" approach, I believe we did it so we could do some limited metaprogramming on the SystemVerilog implementations, right? But I believe we now use SV `parameter`s for all those compile-time parameters… i.e., we rely on SV itself for metaprogramming rather than doing it ourselves. So the need has diminished. Does this sound about right?

rachitnigam, 2020-12-23:
To be frank, I don’t really remember why we used it in the first. I know that the interface needs to know the parameters but beyond that, we don’t really need the Verilog string AFAICT.

sampsyo, 2020-12-23:
Right. You could imagine a different design where we actually have to interpolate the parameters into the Verilog string, but fortunately Verilog `parameter` avoids the need for that.

PR #325 (2020-12-25): Extern and import implementation
Fixes #323. Fixes #171. Fixes #268.

Introduces the `extern`:
```
extern "foo.sv" {
  // Defined in foo.sv
  primitive bar(in: 32) -> (out: 32);
}
```

There is no magical "futil library" anymore. Everything is just a FuTIL file. The `primitives/std.lib` is just a bunch of `import`s into files that define `extern`s. 

The compiler generates a "fat" verilog file by copying the contents of every `extern` file. In the future, we can consider turning some of them into `include` statements.
closed 2020-12-25 by rachitnigam